AlexandreRouma/SDRPlusPlus #767: linux heavy system load

luisteraar, 2022-05-26:
on ubuntu with gnome
srdpp processor load 168%
gnome processor load 131 %
To high gui refresh rate ?
Resulting in alsa output buffer underruns

phrankinstein, 2022-06-19:
Similar issue myself. Are you running this on bare metal or VM?

Proxmox 7.2 Host - Intel i7-7700T @ 2.9Ghz
VM Guest Running on Debian 10 with xfce - 4GB Ram, 4CPU Cores
Open SRP++ Application without starting the radio
Htop output sorted by CPU:
sdrpp 300% CPU Load
sdrpp 73% CPU Load
sdrpp 73% CPU Load
sdrpp 72% CPU Load
sdrpp 70% CPU Load

Debug info showing 15+ms/frame. Framerate ~60FPS Idle. 

Not able to get Intel integrated graphics passthrough working for OpenGL if anyone has any suggestions.

AlexandreRouma, 2022-06-22:
No real opengl = having to render the entire UI on the CPU. It's gonna be slow.

closed 2022-06-22 by AlexandreRouma