AlexandreRouma/SDRPlusPlus #771: Problem with Debian sid release v1.0.4 package

btorrem, 2022-05-28:
There is a problem with the SDR++ v1.0.4 debian sid package sdrpp_debian_sid_amd64.deb

$ sdrpp
sdrpp: error while loading shared libraries: libvolk.so.2.4: cannot open shared object file: No such file or directory

$ apt-cache policy libvolk2-dev
libvolk2-dev:
  Installed: 2.5.1-2
  Candidate: 2.5.1-2
  Version table:
 *** 2.5.1-2 500
        500 http://deb.debian.org/debian unstable/main amd64 Packages
        100 /var/lib/dpkg/status

AlexandreRouma, 2022-05-28:
you're on debian SID, a version of debian that changes its, library versions constantly. The release is over 8 month old, it's not gonna work.

If you're on debian SID (which is a rolling release distro), I assume you're familiar with building software from source, so you're gonna have to go that route.

closed 2022-05-28 by AlexandreRouma

btorrem, 2022-05-30:
Hello Alexandre, thanks for your quick response.
Indeed I'm running Debian Sid (Siduction, Aptosid, Sidux) already for over 15 years, but I can only remember once compiling a package from source. I'm certainly not familiar with building software from source and probably won't go that way for SDR++.. Debian Sid is more stable than most people think, with the advantage of having a rolling release and quite up-to-date software packages.
For SDR++, I think the "hard" dependency problems can be avoided, even for a moving target like Debian Sid.

AlexandreRouma, 2022-05-30:
There is no way around the "hard dependency" issue. All libraries have a specific `soname`, usually tied to there version. Any time you update a library to a new version, the soname changes and it's like the library isn't even installed.
The only way for software to keep working is for it to be updated along with the libraries to be up to date.
This is how all software on rolling release distros work.

For software that doesn't have the luxury of being in the package manager and being automatically rebuilt for new versions,
The software maintainer has to do all the work and constantly rebuilt their software.
Considering I'm not a debian user, I don't do that.
Just like on Archlinux, the only way when not in the package manager is to build from source.