AlexandreRouma/SDRPlusPlus #772: Kernel_task reached 400% when starting up SDR++

ja-Bertolin, 2022-05-28:
Since I updated MacOS BigSur to 11.6.6 I am facing with #689 version of SDR++ high CPU  (300%- 400%) with kernel_task process

Still testing different combinations but so far this is clear, what could I do it?

73

ja-Bertolin, 2022-05-29:
Just as an update: I have realized that it happens after a screensaver activation. Few minutes ago it happened. I stopped (closed) SDR++ and cpu load returned to normal values. I have just restarted and now it seems that it is fine (around 10% CPU) . In some point of time it spikes to 400%.

ja-Bertolin, 2022-06-11:
Just as an update: After several tests, Kernel_task in MacOS gets more than 300% of CPU just when using SDR++, sometimes few minutes after running but it reaches too high CPU. I will try to get an older version of SDR++ to understand from which version it has such behaviour. Once you close SDR++ (stopping it does not fix the problem), CPU comes back to normal load (kernel_taks below 10%). Something is happening during the running of SDR++ (I do not remember such behaviour in the past).

@AlexandreRouma do you recommend me to start from any specific version? any specific change that you could think that it could provoke it?

On the other hand, with the current version #689 the problem of crashing in MacOS does not happen anymore. 

This is the reason that I pointed out how useful is being aware of the changes for every new nightly version in order to be sure potential weird behaviours. Including release notes would be great.

ja-Bertolin, 2022-06-11:
Trying to  from 653 version but crashing all the time. The unique that starts up fine is 673 but the kernek_task load increases quickly reaching more that 250% CPU load-

ja-Bertolin, 2022-06-11:
Update: Discovering that the problem occurs when connected external monitor. If you disconnect the external monitor Kernel_task is below 10% with SDR++ running. If you connect the monitor after starting up SDR++ it seems that kernel_task % increases to 20% but keeps the number around that. When I had connected the monitor and starting up SDR++ the % jumped to up 200-400% or more

ja-Bertolin, 2022-06-12:
Last update: I was wrong, now I do not have the external monitor connected and just started up the SDR++ and Kernek_task got 400%... no way, something happens with the SDR++ that provokes the control of Kernel_task to reduce the impact on CPU.

ja-Bertolin, 2022-06-24:
Again: With the last versions (higher than 716) kernel_task is increasing CPU consumption just when starting up SDR++ reaching in seconds 400% or more. According to the explanation of the process kernel_task in MacOS, it increments CPU when MacOS detects an exaggerated consumption of resources by app that provokes the increase of temperature. In order to avoid burning the circuits, kernel_task takes CPU to reduce the overload of the app guilty of the problem and it slowdowns all the processes reaching 500% CPU.

Observing SDR++, I suspect that it is taking too much resources (just starting up) and that is the reason, it increments the temperature and kernel_taks aims to stop it. With 716 I bought a new 3 fans equipment to reduce the temperature of MacOs and I was able to use 716 much better but now, even using the 3 external fans it is not able to control de T of the MacBook.

I am aware that @AlexandreRouma do not have MacBook to test but in order to make the app useable, we need to help you in some way.  If interested, I could help you to test the app in Mac and provide you parameters, values, etc.. in order to make more efficient the programming of the app for MacOs. Just let me know.

Best and thanks for the great job/effort done so far to provide us a strong SDR app, for the time being the best for MacOS

AlexandreRouma, 2024-01-22:
Was never able to replicate this on my 2009 mac pro and nobody else has ever reported this issue so I'll assume this was a problem with that specific computer or install and close the issue.

closed 2024-01-22 by AlexandreRouma