AlexandreRouma/SDRPlusPlus #774: Signal to Noise not visible on small screens

kencormack, 2022-05-30:
High praise, to the author and contributors.  SDR++ is running great on my Pi 3B+ (Raspberry Pi "Bullseye 32-bit).

There is, however, an issue I'd love to see corrected.  This Pi is using the official Raspberry Pi 7" Touch Screen for it's display.  The screen is 800x480 pixels in size.  On such a small screen, the Signal to Noise graph is compressed, obscured, overwritten by the Icon, or whatever, and is unusable.  See the attached screenshot for reference.

Perhaps the volume slider could be re-worked, to make more usable room available in the Top Bar area?  Making it smaller, making it an auto-hiding control, or even placing it on top of the Left Menu where it could be called upon when needed (but otherwise kept out of sight) would be possibilities.  This last would be my personal favorite, given that on a small touch screen, I'd still need the slider to be big enough to control with a touch/fingertip.  Regardless, the screenshot shows just how large a percentage of horizontal space it occupies on the Top Bar, versus the width of the display.

[image]
.

bubbaearle, 2022-06-16:
I am also using that same screen. Change your resolution to 1024 x 624.

Great screen, crappy default resolution.

kencormack, 2022-06-17:
> I am also using that same screen. Change your resolution to 1024 x 624.

For older eyes, that's not a very practical solution, unfortunately.  (It also scales icons and such, that fat fingers still need to touch accurately.)

bubbaearle, 2022-06-17:
I feel your pain....

This is my set up. I built the Pi unit to be a self contained portable
unit, then I acquired the laptop for free at work.

Follow the link below for more details.

GROUP DESIGN PROJECT - A portable SDR unit.

kencormack, 2022-06-17:
> Reckon a soft tip stylus would help clicking on the touchscreen? Maybe. #2 pencil eraser??

I'll probably just ignore the volume control that I never use (this pi doesn't need/have a speaker), and forget about the StoN scale that I can't see.

bubbaearle, 2022-06-17:
Gotcha

I use a small keyboard with a track pad on mine. Bluetooth speaker for audio.