AlexandreRouma/SDRPlusPlus #778: [Android] v.1.1.0 with Airspy stops after few seconds

catch220, 2022-06-03:
Hello!

Android 9.0, sdr++ v.1.1.0. App recognizes AirSpy (cool!!!) and starts playing FM stations (outstanding!!). But it totally stops after few seconds (1-3 sec in average). Sometimes it can play FM for 20-30 seconds. "Play" button remains switched on, but no sound and no spectrum changes. In Debug section buffer = 0 in any case. I can change central frequency when the app plays sound. When I change to 10 MSPS, it rarely play more than 1-2 sec. Decimation can't change things, any other option can't change things.

Can you possibly help?

AlexandreRouma, 2022-06-03:
bad contact on the USB cable

catch220, 2022-06-03:
Checked. Sorry, but no... I've used test apk from https://github.com/demantz/airspy_android , and there was no any problems in output file in 3 minutes rx

catch220, 2022-06-03:
Finally, got it. It was strong jammer in 446M nearby. Laptop was ok with it, but tablet is too sensitive. I've made very small cable with ring, and things are rolling on)) all in all, thanks for idea!

Have you any plans to add DSD+ extention? Or can I add it by myself somehow?

AlexandreRouma, 2022-09-13:
digital speech support is planned but there is no ETA.

closed 2022-09-13 by AlexandreRouma