AlexandreRouma/SDRPlusPlus #783: [Windows 7 x64] wont load up or start app tried nighty build but same

francisuk1989, 2022-06-13:
Tired installing the latest visual studio but wont work, I have also tried the nighty version but same.
I can see it creates a config.json but after that, nothing.

AlexandreRouma, 2022-06-22:
Run the nightly with `-c` and show what it says.
I'm probably gonna guess broken graphics drivers.
Windows 7 is extremely out of date anyway and might not continue to be supported in the future

francisuk1989, 2022-06-26:
> Run the nightly with `-c` and show what it says. I'm probably gonna guess broken graphics drivers. Windows 7 is extremely out of date anyway and might not continue to be supported in the future

Please dont drop Windows 7 support yet as my issue is the OpenGL drivers on my laptop as is a very old 2006 XP with only OpenGL1,4 is the max support for Windows sadly (Intel GMA 950) with a dual core CPU.

However switching to a Linux OS will give me OpenGL 2.1 just sadly WIndows is only 1.4 max.

closed 2022-06-26 by francisuk1989