AlexandreRouma/SDRPlusPlus #784: SRDpp rigctl set mode not working with terminal emulator

M0XFX, 2022-06-14:
Hi I am hoping to write some code to control the tuning of SDRpp for a radio project however; I can’t seem to be able to remotely change the mode setting.  
I am using v1.0.4 64 bit running on win-10, I have the Rigctl server set to tuning, the local host and port 4532 and the IP is 127.0.0.1. Also I am using RTLSDR dongle.
For testing I am using Telnet (Not in encryption mode) on the local machine to connect and for the most part it all seems to be working. 
I can send (f, get_freq) or (f) and it returns the frequency in Hz Also I can send F, (frequency in Hz) and it will change frequency no problem however if I use the set_freq command it will crash SDR++.
I can also send (m, get_mode) or (m) and it returns the mode and passband setting but I am unable to change the mode setting using any of the following commands. (M, LSB)(M, LSB, 3000)(M, set_mode ‘LSB’’3000’) Note I am just using LSB as an example it is the same on all modes. I have tried lots of different combinations without success. And the terminal returns RPRT 1 in all cases. I have looked at the SDR++ Rigctl main.cpp and it looks like it should work.
Can anyone please shed some light as to where I am going wrong? Thank you in advance.

AlexandreRouma, 2022-06-22:
Likely a bad command synthax, it works fine for me on every device I have. Make sure you're using pre hamlib 4.4 syntax as that one contains breaking changes

M0XFX, 2022-06-24:
Hi Alex, Thank you for getting back to me,  I tried hamlib 4.4 and yes it works without any problems. I need to figure out the correct syntax. I sniffed the comms and it looks the same as what I am sending from a telnet terminal em'. I am guessing that you have incoperated the hamlib libary into SDR++. I will look into the hamlib code and I will leave an update if I figure it out.   regards John

AlexandreRouma, 2022-09-13:
closing as idle

closed 2022-09-13 by AlexandreRouma