AlexandreRouma/SDRPlusPlus #791: clang warnings

yurivict, 2022-06-23:
clang-13 prints many warnings:
```
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/gui/menus/vfo_color.cpp:2:
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/gui/gui.h:7:
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/gui/main_window.h:6:
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/signal_path/vfo_manager.h:2:
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/dsp/vfo.h:4:
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/dsp/resampling.h:115:13: warning: 'calcOutSize' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
        int calcOutSize(int in) {
            ^
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/dsp/vfo.h:125:39: note: in instantiation of template class 'dsp::PolyphaseResampler<dsp::complex_t>' requested here
        PolyphaseResampler<complex_t> resamp;
                                      ^
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/dsp/block.h:68:21: note: overridden virtual function is here
        virtual int calcOutSize(int inSize) {
                    ^
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/gui/menus/vfo_color.cpp:4:
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/signal_path/signal_path.h:2:
In file included from /disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/signal_path/dsp.h:2:
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/dsp/routing.h:169:13: warning: 'run' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
        int run() {
            ^
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/signal_path/dsp.h:57:35: note: in instantiation of template class 'dsp::Reshaper<dsp::complex_t>' requested here
    dsp::Reshaper<dsp::complex_t> reshape;
                                  ^
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/dsp/block.h:73:21: note: overridden virtual function is here
        virtual int run() = 0;
                    ^
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/gui/menus/vfo_color.cpp:133:25: warning: format string is not a string literal (potentially insecure) [-Wformat-security]
            ImGui::Text(name.c_str());
                        ^~~~~~~~~~~~
/disk-samsung/freebsd-ports/comms/sdr++/work/SDRPlusPlus-1.0.4/core/src/gui/menus/vfo_color.cpp:133:25: note: treat the string as an argument to avoid this
            ImGui::Text(name.c_str());
                        ^
                        "%s", 
3 warnings generated.
```

AlexandreRouma, 2022-06-23:
It's all useless warnings that only clang complains about.

shuyuan-liu, 2022-06-24:
I doubt the people making Clang would add these warnings if they were just "useless". @AlexandreRouma I'm not experienced enough to tell you whether they are good advice to follow but please do consider their meaning before dismissing them as useless. If you have done so it would be appreciated if you could explain in a bit more detail why you choose to not explicitly mark functions as `override` and why you don't think the security issue is important.

AlexandreRouma, 2022-06-24:
I did read the warnings.
The one about overrides just complain that I did not explicitely specify the function overrides the virtual function of the parent class.
This is not mandatory and the only reason it complains is because some other embedded code (either imgui or nholmann json) uses it (hence why the warning is "**inconsistent**-missing-override". In my opinion the override specifier for this type of inheritence is visual bloat more than anything and just makes the code uglier.
The format security thing just complains that ImGui can expect a format string as the input of ImGui::Text.
Since the name parameter is only ever controlled by the local user, it's a none issue.
Text fields that display network data have all be properly setup to not parse as format strings.

AlexandreRouma, 2022-06-24:
I will add that neither GCC not MSVC with warnings enable care about any of these. Clang is known to be quite verbose and complain about almost anything

yurivict, 2022-06-24:
Having ```override```/```final``` keywords has the benefit of defining polymorphic behavior cleanly and prevent certain bugs.

AlexandreRouma, 2022-06-24:
In my opinion, not worth the eye sore and the trouble. The less to specify, the cleaner and more readable the code becomes.

gvanem, 2022-06-24:
BTW, the Qt headers are full of such warning suppressor for clang-cl. E.g. a pre-processed output of `class QObject`.
```c
class __declspec (dllimport) QObject
{
  public:
# 129 "f:/gv/Qt/5.15.2/msvc2019/include/QtCore/qobject.h" 3
#pragma clang diagnostic push
# 129 "f:/gv/Qt/5.15.2/msvc2019/include/QtCore/qobject.h" 3
#pragma clang diagnostic ignored "-Winconsistent-missing-override"
# 129 "f:/gv/Qt/5.15.2/msvc2019/include/QtCore/qobject.h" 3
    static const QMetaObject staticMetaObject;
 ...
# 129 "f:/gv/Qt/5.15.2/msvc2019/include/QtCore/qobject.h" 3
#pragma clang diagnostic pop
...
};
```

yurivict, 2022-06-24:
I have 25+ years of C++ use in all possible projects, and I don't agree with such dismissive attitude.
These warnings are generally good since they help to prevent bugs.