AlexandreRouma/SDRPlusPlus #793: WIndow fails to open: Missing OpenGL version

yurivict, 2022-06-23:
I wasn't able to run SDR++ because ```glewInit()``` fails with ```Missing GL version``` error.

This patch helped to reveal this error:
```
--- core/src/core.cpp.orig      2022-06-23 16:55:03 UTC
+++ core/src/core.cpp
@@ -383,8 +383,10 @@ int sdrpp_main(int argc, char *argv[]) {
         free(icons[i].pixels); 
     }
 
-    if (glewInit() != GLEW_OK) {
+    GLenum err = glewInit();
+    if (err != GLEW_OK) {
         spdlog::error("Failed to initialize OpenGL loader!");
+        spdlog::error(glewGetErrorString(err));
         return 1;
     }
```

My video card might be too old - it supports OpenGL-3.0:
```
$ sdrpp 
[2022-06-23 11:58:48.902] [info] SDR++ v1.0.4
[2022-06-23 11:58:48.903] [info] Loading config
[2022-06-23 11:58:48.903] [warning] ConfigManager locked, waiting...
[2022-06-23 11:58:49.072] [info] Using OpenGL 3.0
[2022-06-23 11:58:49.247] [error] Failed to initialize OpenGL loader!
[2022-06-23 11:58:49.247] [error] Missing GL version
```

Is low supported OpenGL version a problem, or something else is causing this?
You might want to add a version check and also to report error details like the above patch does.

AlexandreRouma, 2022-06-23:
you're using an over 8 month old release. GLEW was dropped in favor of Dear ImGui's built in GLAD loader. Use the latest master commit instead

yurivict, 2022-06-23:
Current master also fails:
```
$ sdrpp 
[2022-06-23 12:57:21.276] [info] SDR++ v1.0.6
[2022-06-23 12:57:21.276] [info] Loading config
[2022-06-23 12:57:21.277] [warning] ConfigManager locked, waiting...
[2022-06-23 12:57:21.450] [info] Using OpenGL 3.0
[2022-06-23 12:57:21.730] [info] Loading icons
[2022-06-23 12:57:21.739] [info] Loading band plans
[2022-06-23 12:57:21.743] [info] Loading band plans color table
[2022-06-23 12:57:21.750] [info] Loading modules
[2022-06-23 12:57:21.750] [info] Loading /usr/local/lib/sdrpp/plugins/audio_sink.so
[2022-06-23 12:57:21.756] [info] Loading /usr/local/lib/sdrpp/plugins/discord_integration.so
[2022-06-23 12:57:21.757] [info] Loading /usr/local/lib/sdrpp/plugins/file_source.so
[2022-06-23 12:57:21.761] [info] Loading /usr/local/lib/sdrpp/plugins/frequency_manager.so
[2022-06-23 12:57:21.763] [warning] ConfigManager locked, waiting...
[2022-06-23 12:57:21.763] [info] Loading /usr/local/lib/sdrpp/plugins/hackrf_source.so
[2022-06-23 12:57:21.766] [info] Loading /usr/local/lib/sdrpp/plugins/meteor_demodulator.so
[2022-06-23 12:57:21.769] [info] Loading /usr/local/lib/sdrpp/plugins/network_sink.so
[2022-06-23 12:57:21.771] [info] Loading /usr/local/lib/sdrpp/plugins/plutosdr_source.so
[2022-06-23 12:57:21.777] [info] Loading /usr/local/lib/sdrpp/plugins/radio.so
[2022-06-23 12:57:21.786] [info] Loading /usr/local/lib/sdrpp/plugins/recorder.so
[2022-06-23 12:57:21.790] [info] Loading /usr/local/lib/sdrpp/plugins/rfspace_source.so
[2022-06-23 12:57:21.793] [info] Loading /usr/local/lib/sdrpp/plugins/rigctl_server.so
[2022-06-23 12:57:21.803] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-06-23 12:57:21.806] [info] Loading /usr/local/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-06-23 12:57:21.808] [info] Loading /usr/local/lib/sdrpp/plugins/sdrpp_server_source.so
[2022-06-23 12:57:21.811] [info] Loading /usr/local/lib/sdrpp/plugins/soapy_source.so
[2022-06-23 12:57:21.814] [info] Loading /usr/local/lib/sdrpp/plugins/spyserver_source.so
[2022-06-23 12:57:21.816] [info] Loading /usr/local/lib/sdrpp/plugins/airspy_source.so
[2022-06-23 12:57:21.819] [info] Initializing Airspy Source (airspy_source)
[2022-06-23 12:57:21.820] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-06-23 12:57:21.821] [error] Module 'airspyhf_source' doesn't exist
[2022-06-23 12:57:21.821] [info] Initializing Audio Sink (audio_sink)
[2022-06-23 12:57:21.822] [info] Initializing BladeRF Source (bladerf_source)
[2022-06-23 12:57:21.824] [error] Module 'bladerf_source' doesn't exist
[2022-06-23 12:57:21.824] [info] Initializing File Source (file_source)
[2022-06-23 12:57:21.825] [info] Initializing Frequency Manager (frequency_manager)
[2022-06-23 12:57:21.826] [info] Initializing HackRF Source (hackrf_source)
[2022-06-23 12:57:21.833] [info] Initializing LimeSDR Source (limesdr_source)
[2022-06-23 12:57:21.836] [error] Module 'limesdr_source' doesn't exist
[2022-06-23 12:57:21.836] [info] Initializing Network Sink (network_sink)
[2022-06-23 12:57:21.839] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-06-23 12:57:21.841] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-06-23 12:57:21.842] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-06-23 12:57:21.844] [info] Initializing Radio (radio)
W: [(null)] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

W: [(null)] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.

RtApiPulse::DeviceInfo pa_context_connect() failed: Connection refused

Segmentation fault
```

PulseAudio works on this compurer.

AlexandreRouma, 2022-06-24:
Seems like an issue to report to the RtAudio developpers, SDR++ doesn't handle any audio protocol directly.
If the segfault happens in the SDR++ code (I doubt) I can mitigate it and just abort the loading of the module but otherwise, I have no control over it.

mitchy93, 2022-07-06:
install these packages manually
https://packages.debian.org/bullseye/libvolk2.4
https://debian.pkgs.org/11/debian-main-amd64/libglew2.1_2.1.0-4+b1_amd64.deb.html (**arm64 version also in this url**)

it opened fine after that

AlexandreRouma, 2022-07-06:
NO NO NO.
Don't install any deb package like that. Changing the library version your distro expects will just break things.
The issue literally has nothing to do with glew or volk

mitchy93, 2022-07-06:
This is a temporary work around for the issue that actually works until the references to the older library versions in the code are updated for Ubuntu 22.04

closed 2022-08-17 by AlexandreRouma