AlexandreRouma/SDRPlusPlus #794: Display Issue macOS

Pytlicek, 2022-06-23:
**OS:** macOS Monterey 12.4 on MacBook Pro 16" 2019
**Build used:** https://github.com/AlexandreRouma/SDRPlusPlus/actions/runs/2545681172

**Description:**
You can horizontally scroll the left sidebar by 2-fingers gesture on macOS and you will see `AGC Decay` text on a blank background, as you can see on the screenshot below. YT video: https://youtu.be/IvYooflQOLg

[image]

AlexandreRouma, 2022-06-23:
Looks like  forgot to remove the right text, will push a fix

AlexandreRouma, 2022-06-23:
fixed

closed 2022-06-23 by AlexandreRouma

Pytlicek, 2022-06-23:
Thanks. It works as expected now 👍