AlexandreRouma/SDRPlusPlus #796: Required package not found (lglew)

on4aol, 2022-06-28:
OS  : Linux Ubuntu 20.04 flavour ( Linux Lite 5.8)
Building from source  SDRPlusPlus-1.04 with brew and cmake 

By command cmake i get this error
---------------------------------------------------------------------------

cmake ..   -DOPT_BUILD_AUDIO_SINK=OFF   -DOPT_BUILD_RTL_SDR_SOURCE=ON   -DOPT_BUILD_RTL_TCP_SOURCE=ON   -DOPT_BUILD_SOAPY_SOURCE=OFF
-- Checking for module 'glew'
--   No package 'glew' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  core/CMakeLists.txt:71 (pkg_check_modules)

---------------------------------------------------------------------------

glew was installed by the brew install command .

When i try to install it on the commandline with apt , i see that libglew-dev is already installed 

sudo apt-get install -y libglew-dev
Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd       
De statusinformatie wordt gelezen... Klaar

libglew-dev is reeds de nieuwste versie (2.1.0-4).  ( newest version )

Also installed glew- utils with no success.

Any hints ? 
on4aol , Luc

AlexandreRouma, 2022-06-28:
Why did you install brew with glew? You're on linux, you need to use your distro's package manager (in this case apt)

on4aol, 2022-06-28:
Well,
I tried to install it via the .deb package on the commandline and i get this warning

 on4aol  ~  2  sudo dpkg -i sdrpp_ubuntu_focal_amd64.deb
Voorheen niet geselecteerd pakket sdrpp wordt geselecteerd.
(Database wordt ingelezen ... 453409 bestanden en mappen momenteel geïnstalleerd.)
Uitpakken van sdrpp_ubuntu_focal_amd64.deb wordt voorbereid...
Bezig met uitpakken van sdrpp (1.0.4-424) ...
Instellen van sdrpp (1.0.4-424) ...
Bezig met afhandelen van triggers voor desktop-file-utils (0.24-1ubuntu3) ...
Bezig met afhandelen van triggers voor gnome-menus (3.36.0-1ubuntu1) ...
Bezig met afhandelen van triggers voor mime-support (3.64ubuntu1) ...
 on4aol  ~  sdrpp

______________________________________________________________________
sdrpp: error while loading shared libraries: libvolk.so.2.2: cannot open shared object file: No such file or directory

___________________________________________________________________________
error on libvolk.

So , later on i tried with brew , but that was a mistake of myself . now i see it that it is for MACOS.

So , still no cigar 
Any comment on this new error ?

AlexandreRouma, 2022-06-28:
Are you sure you are ubuntu focal? Did you install every listed dependency using apt?

on4aol, 2022-06-28:
Ok Alexandre,
Thanks for the effort.

My usr/local/lib  contains the following 

 on4aol  /  usr  local  lib  ls -al
total 13420
drwxr-xr-x  7 root root     4096 mei 27 10:21 .
drwxr-xr-x 10 root root     4096 jan 20  2020 ..
drwxr-xr-x  3 root root     4096 okt 29  2021 cmake
-rw-r--r--  1 root root  8294566 mei 27 10:05 libliquid.a
-rw-r--r--  1 root root  4439456 mei 27 10:05 libliquid.so
-rw-r--r--  1 root root   113866 okt 29  2021 librtlsdr.a
lrwxrwxrwx  1 root root       14 okt 29  2021 librtlsdr.so -> librtlsdr.so.0
lrwxrwxrwx  1 root root       19 okt 29  2021 librtlsdr.so.0 -> librtlsdr.so.0.6git
-rw-r--r--  1 root root    92072 okt 29  2021 librtlsdr.so.0.6git
-rw-r--r--  1 root root   468790 mei 27 10:21 libsoundio.a
lrwxrwxrwx  1 root root       15 mei 27 10:21 libsoundio.so -> libsoundio.so.2
lrwxrwxrwx  1 root root       19 mei 27 10:21 libsoundio.so.2 -> libsoundio.so.2.0.0
-rw-r--r--  1 root root   292088 mei 27 10:21 libsoundio.so.2.0.0
drwxr-xr-x  2 root root     4096 okt 29  2021 pkgconfig
drwxrwsr-x  4 root staff    4096 mrt 18  2021 python2.7
drwxrwsr-x  3 root staff    4096 jan 20  2020 python3.7
drwxrwsr-x  3 root staff    4096 jan 20  2020 python3.8

There are a few logical links for other   .so    but i don't see something for   libvolk.so.2.2

I have another desktop and also a older laptop with the same OS.
I'll give it a try and let you know if i get the same problems.

Thanks ,
Luc ON4AOL

on4aol, 2022-06-28:
Goodevening Alexandre
I did a new install under my other desktop with the same OS:
Well , i do have other results , but is now working.
Let me explain what this time the problem was 

  Pakket libglew-dev is niet geïnstalleerd. ( not installed)

Then i manually installed with follow command

sudo apt install libglew-dev

and then this message came  through

De volgende pakketten hebben niet-voldane vereisten:
(The following packages have unfulfilled requirements:)

 libglew-dev : Vereisten: libglew2.1 (= 2.1.0-4)  will not be installed
Prerequisites: libglu1-mesa-dev but it will not install of
 libglu-dev

There are unfulfilled requirements. It is best to run 'apt --fix-broken install' without specifying any packages, (or you can specify a fix yourself).

So  I ran   apt --fix-broken install

( it's in Dutch , sorry about that but there are now 3 new packages installed ( libglew-dev libglew2.1 libglu1-mesa-dev)
sudo apt --fix-broken install
Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd       
De statusinformatie wordt gelezen... Klaar
Vereisten worden gecorrigeerd... Klaar
De volgende extra pakketten zullen geïnstalleerd worden:
  libglew-dev libglew2.1 libglu1-mesa-dev
Voorgestelde pakketten:
  glew-utils
De volgende NIEUWE pakketten zullen geïnstalleerd worden:
  libglew-dev libglew2.1 libglu1-mesa-dev
0 opgewaardeerd, 3 nieuw geïnstalleerd, 0 te verwijderen en 7 niet opgewaardeerd.
1 niet volledig geïnstalleerd of verwijderd.

And final with success
sudo dpkg -i sdrpp_ubuntu_focal_amd64.deb

I hope this will maybe usefull for others too

[image]

mitchy93, 2022-07-06:
install these packages manually
https://packages.debian.org/bullseye/libvolk2.4
https://debian.pkgs.org/11/debian-main-amd64/libglew2.1_2.1.0-4+b1_amd64.deb.html (**arm64 version also in this url**)

it opened fine after that

closed 2022-08-17 by aosync