AlexandreRouma/SDRPlusPlus #803: SpyServer Autostart crash

gvanem, 2022-07-10:
I first had the SpyServer module active on the previous run of `sdrpp.exe`.
The starting it again with `sdrpp.exe --con --autostart --root .\gv-config`, the
`modules/spyserver_source.dll` crashed on a NULL-ptr access:
```
spyserver_source!spyserver::SpyServerClientClass::sendCommand
mov eax,dword ptr [ecx+38h] ds:002b:00000038=????????  ECX == 0
```

Call-stack:
```
00 spyserver_source!spyserver::SpyServerClientClass::sendCommand(void)
01 spyserver_source!spyserver::SpyServerClientClass::setSetting(unsigned int setting = 0x64, unsigned int arg = 1)
02 spyserver_source!SpyServerSourceModule::start(void * ctx = 0x10774f78)+0x39
03 sdrpp_core!SourceManager::start(void)+0xf
04 sdrpp_core!MainWindow::setPlayState(bool _playing = true)+0x29
05 sdrpp_core!MainWindow::draw(void)+0x6ca
06 sdrpp_core!backend::renderLoop(void)+0x664
07 sdrpp_core!sdrpp_main(int argc = 5, char ** argv = 0x00d30628)+0x374c
...
```

Seems it sends the START command before it's event connected.
So I patched it like this:
```diff
--- a/source_modules/spyserver_source/src/main.cpp 2022-07-08 19:44:03
+++ b/source_modules/spyserver_source/src/main.cpp 2022-07-10 15:30:29
@@ -119,7 +119,7 @@

     static void start(void* ctx) {
         SpyServerSourceModule* _this = (SpyServerSourceModule*)ctx;
-        if (_this->running) { return; }
+        if (_this->running || !_this->client) { return; }

         int srvBits = streamFormatsBitCount[_this->iqType];
         _this->client->setSetting(SPYSERVER_SETTING_IQ_FORMAT, streamFormats[_this->iqType]);
```

This worked w/o a crash. But I'm not sure it's correct.

AlexandreRouma, 2022-07-10:
hmm I'm not actually sure what the best way to fix this is. this way would keep it incompatible with autostart mode.
I think having it try to connect if not already connected would be a better solution.
I'll have a look a the code.

AlexandreRouma, 2022-07-10:
Should be fixed in the latest commit

closed 2022-07-10 by AlexandreRouma

gvanem, 2022-07-10:
> Should be fixed in the latest commit

Which one? I cannot see any commit the last 18 hours.

AlexandreRouma, 2022-07-10:
seems like the internet connection had dropped out, should be up now

gvanem, 2022-07-10:
>  should be up now

Applied the commit. Works fine. Thanks!