AlexandreRouma/SDRPlusPlus #806: How To Build On Fedora

BT-Justice, 2022-07-11:
How do you build SDR++ on Fedora?  There appear to be no Fedora versions of libfftw3-dev libglfw3-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev:

```
$ sudo dnf install libfftw3-dev libglfw3-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev
No match for argument: libfftw3-dev
No match for argument: libglfw3-dev
No match for argument: libvolk2-dev
No match for argument: libsoapysdr-dev
No match for argument: libairspyhf-dev
No match for argument: libiio-dev
No match for argument: libad9361-dev
No match for argument: librtaudio-dev
No match for argument: libhackrf-dev
Error: Unable to find a match: libfftw3-dev libglfw3-dev libvolk2-dev libsoapysdr-dev libairspyhf-dev libiio-dev libad9361-dev librtaudio-dev libhackrf-dev
```
The Fedora supplied SDR++ is v1.0.4.  It will not work with any of my Airspy units.  It does work with my RTL-SDRs though.

Any thoughts about having a Flatpak or AppImage option for SDR++?

AlexandreRouma, 2022-09-13:
I can't start giving instructions for every single package manager out there. Just search for the needed packages and use the distro specific names as needed.

closed 2022-09-13 by AlexandreRouma

BT-Justice, 2022-09-30:
Awesome!

https://github.com/AlexandreRouma/SDRPlusPlus/issues/202#issuecomment-1263989279