AlexandreRouma/SDRPlusPlus #809: Rendering of text not OK, and not clearing old content

pe1chl, 2022-07-16:
I wanted to try SDR++ so I installed the nightly build of Jul 15, 2022 on my Debian Buster system.
It has a NVIDIA GT216 [GeForce 210] videocard using the proprietary driver 340.108 (last available driver for that card).
SDR++ reports:
[2022-07-16 09:40:04.869] [info] Using OpenGL 3.0

The basic window is drawn OK but the text appears smudgy, and when I do something in the menu that causes it to change, the previously written text remains so the column becomes more and more a white blob.
However, the waterfall works fine.

Other software works fine on the machine, also software that uses OpenGL.  I remember once before I tried a program that did not work at first, but it was fixed when I installed an update for that program.
Is this a known and resolvable issue?  Can I provide more info like a debug log?

AlexandreRouma, 2022-08-17:
Cropping the entire boot log to a single line is not exactly helpful to debug the issue.

The issue is very likely just the GPU not supporting the features SDR++ needs to work properly or broken graphics drivers.

pe1chl, 2022-08-17:
Ok I asked if there was other logs I could provide, as the remainder of the boot log does not seem to contain any info relevant to this problem (it only lists a lot of files it is loaded and some details about what is going on with the RTLSDR I configured, but nothing at all about the graphics side of things).
Other OpenGL software like google-earth works perfectly fine using the same driver, is there some way to check which features are not supported but are required?

AlexandreRouma, 2022-10-04:
It doesn't use anything special. If it's not clearing it means a simple `glClear(GL_COLOR_BUFFER_BIT);` fails. To me that can only mean that either the driver is broken and refuses to clear or it has non-sane defaults which somehow masks every color bit during the clear operation. If that's the case it can be worked around but it's incredibly suspicious and still is a driver compatibility issue. Seeing the age of that GPU I wouldn't be surprised if it's neglected by nVidia.

AlexandreRouma, 2023-03-10:
Assuming this was a driver or hardware issue since nobody else has that issue.

closed 2023-03-10 by AlexandreRouma