AlexandreRouma/SDRPlusPlus #810: MacOS: Kernel_task reached 400% when starting up SDR++, new findings

ja-Bertolin, 2022-07-17:
Testing with different configurations regarding the issue #772  I found, by serendipity, something interesting and I hope it will help to find the problem

I have observed that,if I use speakers of the laptop kernel_task increases up to 400% CPU but if I switch on the headphones, it falls down until regular values (around 20% or less). I have done severals tests and all of them point out that there is a direct relationship with  audio module? it only happens with SDR++

Reminding my condx: MacOS Big Sur 11.6.7, AirSpyHf+

ja-Bertolin, 2022-07-17:
In addition to this, if I run another app like browser and run a Youtube video, I need to close SDR++ , not only to stop it, otherwise kernel_task increase also to 400%. something is inside the SDR++ code that it taking too much resource even though CPU % of SDR++ is less than 100%. Which could be the explanation?

ja-Bertolin, 2022-07-19:
just for your information @AlexandreRouma , tested with GQRX and kernel_task gets values below 15-20%

AlexandreRouma, 2022-07-23:
Again as I've mentioned before, I have no mac and this issue only happens on your system. There is quite literally nothing I can do to know what's wrong and even less I can do to fix it.

closed 2022-07-23 by AlexandreRouma

ja-Bertolin, 2022-07-23:
Ok, thanks. I aware of it, nevertheless, delivering an app for an Operative System which you can not give support as you do not have a Mac to test it, it is not a good practice.

I have just moved to GQRX and getting familiar with its way of working, as SDR++, actually, is not useable for me due to this unsolved problem. If the rest of the mac users do not have such problem with SDR++ great for them, I wish them good luck.