AlexandreRouma/SDRPlusPlus #813: Fails to start on wayland

tabell, 2022-07-22:
Using latest version from Arch AUR and a Wayland-based system (riverwm) I see this on starting
```
√ ~ % sdrpp
[2022-07-22 11:11:51.924] [info] SDR++ v1.0.4
[2022-07-22 11:11:51.925] [info] Loading config
[2022-07-22 11:11:52.022] [error] Glfw Error 65544: Wayland: Focusing a window requires user interaction
[2022-07-22 11:11:52.022] [info] Using OpenGL 3.0
[2022-07-22 11:11:52.059] [error] Glfw Error 65544: Wayland: Setting window icon not supported
[2022-07-22 11:11:52.060] [error] Failed to initialize OpenGL loader!
```

starvald, 2022-07-22:
try installing glfw-x11 instead of glfw-wayland

tabell, 2022-07-22:
Just to be clear I'm not running X11 at all. I tried the suggestion and now it shows this error:

```
√ ~ % sdrpp
[2022-07-22 11:45:00.030] [info] SDR++ v1.0.4
[2022-07-22 11:45:00.030] [info] Loading config
[2022-07-22 11:45:00.030] [error] Glfw Error 65544: X11: The DISPLAY environment variable is missing
```

I tried setting it with `export DISPLAY=:0` but it just says

```
[2022-07-22 11:45:25.996] [error] Glfw Error 65544: X11: Failed to open display :0
```

starvald, 2022-07-22:
v1.0.4 is quite old, trying building sdrpp-git from aur.

tabell, 2022-07-22:
@starvald whatever the problem was, it's fixed in the latest git. thank you kind stranger.

closed 2022-07-22 by tabell