AlexandreRouma/SDRPlusPlus #816: Adding a third "Radio" module segfaults SDR++

RAD750, 2022-07-24:
**Hardware**
- CPU: Intel Core i5-6200u 
- GPU: Intel HD Graphics 520
- Computer: Lenovo Thinkpad x260
- SDR: Local USB RTL-SDR (Nooelec NESDR SMArtee)

**Software**
- Operating System: Pop!_OS 22.04 (based on Ubuntu 22.04), Linux 5.17.15-76051715-generic
- SDR++: 1.1.10 built at 00.40, July 16, 2022

**Bug Description**
Adding a third "Radio" module (VFO) immediately crashes SDR++ with a segmentation fault.

**Steps To Reproduce**
1. Start SDR++
2. Go in Module Manager, find "Radio" module, give it a name like "Radio 3"
3. Click on the "+" button
4. SDR++ crashes.

Attached log of launching `sdrpp`, although I don't see anything

[sdrpp.txt](https://github.com/AlexandreRouma/SDRPlusPlus/files/9175543/sdrpp.txt)

AlexandreRouma, 2022-07-24:
How much ram does the computer have? It's suspicious that it would break at 3 but not at 2.

RAD750, 2022-07-24:
Hi, the computer has 8 GB of DDR4 RAM. 
However, I don't think RAM is the key here. I did some measurements and it still crashes even with 60%+ of free RAM.

Before starting SDR++
```
               total        used        free      shared  buff/cache   available
Mem:            7820        1598        4591         271        1630        5678
Swap:           4095          35        4060
```

After starting SDR++, DSP running
```
               total        used        free      shared  buff/cache   available
Mem:            7820        1722        4393         313        1704        5511
Swap:           4095          35        4060
```

After adding the third Radio module, SDR++ crashed already. At the moment of the crash, there was over 5.5 GB of free memory.
```
               total        used        free      shared  buff/cache   available
Mem:            7820        1617        4564         247        1638        5683
Swap:           4095          35        4060
```

Doing the usual `dmesg | grep "oom"` doesn't show anything regarding the Out of Memory Killer being at work.

Thanks again for your help :)

AlexandreRouma, 2022-10-10:
Only way to figure this out then is running with GDB, but to me I'm almost sure it's an allocation that's failing.
I cna guide you on the discord, not easy to explain in a short message on github.

AlexandreRouma, 2022-10-14:
For those having this issue, the source has been identified and will be fixed soon.