AlexandreRouma/SDRPlusPlus #819: SDR++ Crashes with hackrf plugged in and I click the play button.

AKABigDinner, 2022-07-24:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

AKABigDinner, 2022-07-24:
I am using:
Hardware: Raspberry PI 4 4GB ram.
OS:  Raspberry Pi OS (64-bit).
SDR: Hackrf
Software: SDR++

AlexandreRouma, 2022-07-24:
Are you serious?
1) Duplicate of #760
2) You haven't even filled out the template
3) The fix is [literally the first thing in the troubleshooting section](https://github.com/AlexandreRouma/SDRPlusPlus#sdr-crashes-then-it-wont-start-again-no-matter-what) of the readme, RTFM...
4) "Software: SDR++".... yeah no shit?!  *WHICH VERSION*?
If the stuff I haven't linked isn't the issue, reopen one properly and correctly **fill out** the bug report.

closed 2022-07-24 by AlexandreRouma