AlexandreRouma/SDRPlusPlus #820: SDR++ doesn't start on old laptop with Debian (OpenGl version issue?)

kord1e, 2022-07-25:
**Hardware**
- CPU: Intel i5 520 M
- GPU: 1st gen Intel HD graphics (integrated)
- SDR: local (none)

**Software**
- Operating System: Debian 11 64bit
- SDR++: 1.1.0 built from latest git master branch in July 24

**Bug Description**
SDR++ doesn't start because it doesn't find supported OpenGL version or something.

**Steps To Reproduce**
1. compile SDR++ from source
2. make the configs and start it
3. observe console output

**Only If SDR++ fails to lauch or the SDR fails to start:**
SDR++ seems to start just fine in server mode (-c doesn't work, -s works). GUI mode has this issue.

**Additional info**
Console output on normal start:
```
[2022-07-25 23:00:34.740] [info] SDR++ v1.1.0
[2022-07-25 23:00:34.741] [info] Loading config
[2022-07-25 23:00:34.741] [warning] ConfigManager locked, waiting...
[2022-07-25 23:00:34.774] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[2022-07-25 23:00:34.774] [info] OpenGL 3.0 was not supported
[2022-07-25 23:00:34.778] [error] Glfw Error 65543: GLX: Failed to create context: GLXBadFBConfig
[2022-07-25 23:00:34.778] [info] OpenGL 3.1 ES was not supported
[2022-07-25 23:00:34.810] [info] Using OpenGL 2.1
Failed to initialize OpenGL loader!
[2022-07-25 23:00:34.930] [warning] Could not init using OpenGL with normal GLSL version, falling back to GLSL 1.2
Failed to initialize OpenGL loader!
[2022-07-25 23:00:34.931] [error] Failed to initialize OpenGL with GLSL 1.2
```

AlexandreRouma, 2022-07-25:
integrated GPU is too old. Try updating graphics driver but it likely won't change anything.

closed 2022-07-25 by AlexandreRouma

kord1e, 2022-07-25:
@AlexandreRouma 
I actually tried two X11 drivers (intel and modesetting) as well as wayland but the result was the same...
so does this mean that SDR++ can't be used on old computers like this one?

kord1e, 2022-07-25:
Another question, are there any packages that could help? SDR++ can be compiled just fine so I guess not? So at least basic dependencies are Ok...

AlexandreRouma, 2022-07-25:
If it doesn't support OpenGL 2.1 or GLSL 1.2, it's not supported.
it has nothing to do with any kind of dependencies. The driver either says if it supports the right opengl version at runtime, not compile time.
This is an extremely old computer, I cannot support hardware this old at the cost of performance or features for normal computers.

kord1e, 2022-07-25:
@AlexandreRouma 

found a workaround!

Force software rendering like this:

```
LIBGL_ALWAYS_SOFTWARE=1 sdrpp
```

and SDR++ starts normally. I didn't test much yet but will do later.

AlexandreRouma, 2022-07-25:
seeing how old that CPU is running software rendered is not exactly a usable solution

lwvmobile, 2022-10-27:
> If it doesn't support OpenGL 2.1 or GLSL 1.2, it's not supported.

Just to clarify, what is the current OpenGL requirement for SDR++? Is it GL2.1, GLSL 2.1, or newer. Reason I ask, had a few reports from the RR forums about Windows versions not running, but the console shows SDR++ found 2.1 but requested 3.0.

This screenshot was from the last nightly build.
[image]

AlexandreRouma, 2022-10-27:
that doesn't even try to use 2.1, otherwise it would say it. Either your drivers are broken or your hardware is way too old.

fizr5, 2024-02-08:
> that doesn't even try to use 2.1, otherwise it would say it. Either your drivers are broken or your hardware is way too old.

No, its your broken software that is causing this issue for years and you just ignore it and dont fix the issue or try it out on yourself. You for years dont even check what the hardware supports that the people write down. Because this is first gen Core i series and those support exactly OpenGL 2.1. Its even written for you on Wikipedia. Now fix finally this proven broken software!

kord1e, 2024-02-11:
Another workaround for people who are interested:

Instead of software rendering (see a few posts above) you can try like this

`MESA_GL_VERSION_OVERRIDE=3.0 sdrpp`

and it may or may not work. If it works, it will not bake your CPU like software rendering does. But it's possible that it doesn't work and then software rendering is the only option.