AlexandreRouma/SDRPlusPlus #824: Waterfall resize issue

gvanem, 2022-07-27:
**Hardware**
- CPU:  AMD Ryzen 9 3900X, 3.8 GHz, 12 cores.
- GPU:  NVIDIA GeForce GT1030.
- RAM: 16 GByte.

**Software**
- Windows-10, x64, build 19044.
- SDR++: ver. 1.1.0 built it myself from git master (MSVC-2022).

SDR++ has an issue with `WaterFall::onResize()` and extending / shrinking the waterfall window too quickly. Not sure 
what the *speed-limit* is. E.g. when dragging the waterfall window upwards from this:
[image]

the program could look like:
[image]

This grayed-out User Interface is Windows' way to indicate a "frozen" program. The above "*Drag-line*" is the soon to be top-border of the Waterfall window. This never happens. I added some extra traces in `waterfall.cpp` and some really huge allocations occurs:

```
sdrpp.exe --con --autostart 
...
[2022-07-27 20:41:40.163] [error] line 733: realloc() of 'rawFFT' failed. errno=12, fft_size=372736 kByte
```

Above example was using the SpyServer module connected to k2zn.ddns.net:5555. 
Total mem usage == 1 GByte and 35 threads according to *Process Explorer*.

And dragging the waterfall window *downwards*, could simply abort with a **Security check failure or stack buffer overrun** !!.
The other threads seems to work fine in this hung situation; audio is clear despite a hung UI.

AlexandreRouma, 2022-07-27:
Please include steps to reproduce as asked in the template. I can't replicate here no matter what weird stuff I do to the resize line.
I'm also not sure what you mean by "speed" because the adjustment is not instant. It only resizes once the resize bar is dropped.
Also the log is helpful but I don't know exactly what's being logged here. What is "fft_size"? Why is it in kByte? Which line exactly?
etc

gvanem, 2022-07-27:
> What is "fft_size"? Why is it in kByte? Which line exactly? etc

That's part of my effort to trace this waterfall issue. I'll be back!

gvanem, 2022-07-28:
I forgot to say that the above issue showed itself in 32-bit mode. 

I've now rebuilt SDR++ and the dependencies for 64-bit mode and have difficulty reproducing this out-of-memory issue. Even with a FFT Size of 524288. This is the diff I used;
```diff
--- a/core/src/gui/widgets/waterfall.cpp 2022-07-28 05:16:11
+++ b/core/src/gui/widgets/waterfall.cpp 2022-07-28 06:31:27
@@ -24,6 +24,9 @@
     { 0x4A, 0x00, 0x00 }
 };

+#define RAW_FFT_MEM_FAIL(which, size) \
+        spdlog::error ("line %u: %s (%.1f MByte) for 'rawFFT' failed.", __LINE__, which, size/(1024*1024))
+
 // TODO: Fix this hacky BS

 double freq_ranges[] = {
@@ -710,6 +713,8 @@
         if (waterfallVisible) {
             // Raw FFT resize
             fftLines = std::min<int>(fftLines, waterfallHeight) - 1;
+            size_t fft_size = waterfallHeight * rawFFTSize * sizeof(float);
+
             if (rawFFTs != NULL) {
                 if (currentFFTLine != 0) {
                     float* tempWF = new float[currentFFTLine * rawFFTSize];
@@ -720,10 +725,14 @@
                     delete[] tempWF;
                 }
                 currentFFTLine = 0;
-                rawFFTs = (float*)realloc(rawFFTs, waterfallHeight * rawFFTSize * sizeof(float));
+                rawFFTs = (float*)realloc(rawFFTs, fft_size);
+                if (!rawFFTs)
+                   RAW_FFT_MEM_FAIL("realloc", fft_size);
             }
             else {
-                rawFFTs = (float*)malloc(waterfallHeight * rawFFTSize * sizeof(float));
+                rawFFTs = (float*)malloc(fft_size);
+                if (!rawFFTs)
+                   RAW_FFT_MEM_FAIL("malloc", fft_size);
             }
             // ==============
         }
@@ -1082,13 +1091,20 @@
         std::lock_guard<std::recursive_mutex> lck(buf_mtx);
         rawFFTSize = size;
         int wfSize = std::max<int>(1, waterfallHeight);
+        size_t fft_size = rawFFTSize * wfSize * sizeof(float);
+
         if (rawFFTs != NULL) {
-            rawFFTs = (float*)realloc(rawFFTs, rawFFTSize * wfSize * sizeof(float));
+            rawFFTs = (float*)realloc(rawFFTs, fft_size);
+            if (!rawFFTs)
+               RAW_FFT_MEM_FAIL("realloc", fft_size);
         }
         else {
-            rawFFTs = (float*)malloc(rawFFTSize * wfSize * sizeof(float));
+            rawFFTs = (float*)malloc(fft_size);
+            if (!rawFFTs)
+               RAW_FFT_MEM_FAIL("malloc", fft_size);
         }
         fftLines = 0;
-        memset(rawFFTs, 0, rawFFTSize * waterfallHeight * sizeof(float));
+        if (rawFFTs)
+             memset(rawFFTs, 0, rawFFTSize * waterfallHeight * sizeof(float));
         updateWaterfallFb();
```

But do not now with my new AMD Ryzen 3900X CPU and 16 GByte RAM.  
Previously I hit the issue frequently in 32-mode and 8 GByte (on an Intel-I7 CPU).

So w/o eating a lot more RAM or unplugging a RAM module, I'm not digging further into this.
But the above `if (rawFFTs)` should be added.

*PS*. Using *Process Explorer* while rapidly expanding the Waterfall window, I sometime see this peak:
[image]

before the mem-usage settles again.

AlexandreRouma, 2022-07-28:
Well the 32bit build would have been an important thing to know.
32bit apps can only allocate 2GB of memory on windows (the lower 2gb of the address space are reserved for the kernel).
The issue here is you've got SDR++ probably fullscreen which means the raw FFT buffer used for full waterfall update is HUGE (in addition to the DSP buffers that already use quite a bit of ram).
When resizing, SDR++ has to reallocate those buffers, which means **first** allocating new memory (hence the spike), copy the data and only then free the old memory. This is why you see a crash.

gvanem, 2022-07-28:
Good to know. I guess this is tricky to redesign for a leaner use of RAM. Hence closing.

closed 2022-07-28 by gvanem