AlexandreRouma/SDRPlusPlus #828: Decimation WAV

zeke800, 2022-08-01:
When using the decimation feature to record, SDR++ records at the original sample rate. This is seen during the WAV file playback, when it plays faster than normal. For example, when receiving Meteor (excuse the local VHF noise), the recording was decimated by 4 in SDR++. However, the bandwidth does not even remotely fit. 
[image]

zeke800, 2022-08-03:
Thank you! However, is there a way to convert the existing recordings?

AlexandreRouma, 2022-08-03:
you either have to divide the samplerate and bytesPerSecond fields of the wav header by 4 or use GNU radio to read the wav as raw IQ data and rewrite it with the correct samplerate.

closed 2022-08-03 by AlexandreRouma