AlexandreRouma/SDRPlusPlus #839: Bookmarks export

Terminator0007, 2022-08-10:
Hi. How to export all bookmarks into one file to import in another sdrpp installed ? I see that can only export one bookmark ...

jazzbassNick, 2022-08-14:
You have to select all the bookmarks to export.

Terminator0007, 2022-08-14:
> You have to select all the bookmarks to export.

Thanks, this idea I have tested already but if you want to export 200 bookmarks wich are in 8 groups it is complicated . It should be possible to mark group of bookmarks or all group and export all with one click. Also can be some kind of backup settings which can include settings and bookmarks in one file. If you want to build sdrpp on another computer spend one day to copy all settings and especially 200 bookmarks. If you have more bookmarks in sdrpp take a day off at work or waste your weekend time to do all :-)

jazzbassNick, 2022-08-14:
Try copying over the `frequency_manager_config.json` into the new install

Terminator0007, 2022-08-14:
Dear friend. I don't like make my life harder but always easier.... and do not waste a time for something stupid what can be made by automation... But but .... simple or advanced backup should be available in every software. Therefore I posted enhancement idea in this subject :-)

AlexandreRouma, 2022-08-17:
Adding a button just to do the same thing you would do by copying a file is a bit useless imo. You're wasting more time by clicking the button, saving the file somewhere else when the file literally already exists on the drive... As other users recommended, just copy the `frequency_manager_config.json` file to where you need it.

Terminator0007, 2022-08-17:
> Adding a button just to do the same thing you would do by copying a file is a bit useless imo. You're wasting more time by clicking the button, saving the file somewhere else when the file literally [...]

I understand your point of view and this is not a problem to do it for me but if you install SDRPP on Mac OS everything is more complicated for simple user when SDRPP crushing on Mac OS few times during maintenance. Sometime need to delete app and install again ... One feature more which include all bookmarks, settings for devices I hope it is not a big problem for Author of this super software SDRPP ... I said Super Software because it is only one the Best for Mac M1. Other software avaliable are not on this high level. Why not to improve some features and to be ONLY ONE The Best for Mac OS and kill competitors ? :-)

Terminator0007, 2022-08-17:
> Adding a button just to do the same thing you would do by copying a file is a bit useless imo. You're wasting more time by clicking the button, saving the file somewhere else when the file literally [...]

I understand your point of view and this is not a problem to do it for me but if you install SDRPP on Mac OS everything is more complicated for simple user when SDRPP crushing on Mac OS few times during maintenance. Sometime need to delete app and install again ... One feature more which include all bookmarks, settings for devices I hope it is not a big problem for Author of this super software SDRPP ... I said Super Software because it is only one the Best for Mac M1. Other software avaliable are not on this high level. Why not to improve some features and to be ONLY ONE The Best for Mac OS and "kill" competitors ? :-)

AlexandreRouma, 2022-08-17:
Copying that file on MacOS is not much harder. The nightly `.app` bundle puts its config files in `~/Library/Application Support/sdrpp`.

Also as mentioned in other threads, I develop "blind" for MacOS. I don't own any apple device and I don't have any computer that could run MacOS. This means I also have no way of debugging issues on MacOS.

Terminator0007, 2022-08-17:
> Copying that file on MacOS is not much harder. The nightly `.app` bundle puts its config files in `~/Library/Application Support/sdrpp`.
>
> Also as mentioned in other threads, I develop "blind" for [...]

I remember your comments in other post regarding to Mac OS. It is clear for me but only can say thank you for good job until today as "blind" Author :-) Maybe it is not stupid idea to buy the Mac mini M1 ( because is a good market price ) and develop this app for Mac users. You can sell it in Apple Mac Store with success. I can pay 50EUR for such app if it will be available in AppleStore. And believe me that you can sell a lot for Mac users and make good profit for yourself :-) Price for M1 mini is 800EUR and will return back after 200 buyers. After this you are owner of M1 and the rest of good profit from app sales. :-) Android users are more happy to have SDRPP for this platform but M1 ARM it is not so far to use it for all Mac's or iPad's ... :-) Kind Regards for you and thanks for an answer.

betzburger, 2022-09-09:
> Also as mentioned in other threads, I develop "blind" for MacOS. I don't own any apple device and I don't have any computer that could run MacOS. This means I also have no way of debugging issues on MacOS.

A wonder that it works that well :-) Here on M1 Mac.

AlexandreRouma, 2022-10-03:
> Maybe it is not stupid idea to buy the Mac mini M1 ( because is a good market price ) and develop this app for Mac users.

I don't have the kind of money to buy one used and selling SDR++ is completely unacceptable. When I started out SDR as a kid I would have never been able to afford software like that. I don't want price to hold anyone back from getting a good software experience.

closed 2022-10-03 by AlexandreRouma