AlexandreRouma/SDRPlusPlus #844: Graphics issue with Raspberry Pi 4

srs4511351, 2022-08-16:
**Hardware**
- CPU: Broadcom BCM2711, Quad core Cortex-A72 (ARM v8) 64-bit SoC @ 1.5GHz
- RAM:8GB
- GPU: Integrated
- SDR: SDRplay RSP1, HackRF, RTL-SDR
- OpenGL ES 3.1, Vulkan 1.0

**Software**
- DietPi Debian GNU/Linux 11 (bullseye)
- SDR++:  SDR++ v1.1.0 (Built at 22:49:26, Jul 9 2022)

**Bug Description**
High CPU load when using the default graphics driver.
OpenGL version string: 3.1 Mesa 20.3.5
GLX version: 1.4
SDR++ Debug shows ImGui version 1.87, framerate 25 FPS

Downgrading to an older driver fixes the issue, but this is not a good idea for the future. 
To specify the older driver, in /boot/config.txt, I added these lines:
max_framebuffers=2
dtoverlay=vc4-fkms-v3d
hdmi_enable_4kp60=1

reboot

Now, the version is:
OpenGL version string: 2.1 Mesa 20.3.5
GLX version: 1.4

With older 2.1 driver (vc4-fkms-v3d), the SDR++ terminal output includes:
MESA_GLES_VERSION_OVERRIDE=3.0 MESA_GL_VERSION_OVERRIDE=3.0 sdrpp
[2022-08-16 14:41:08.293] [info] Using OpenGL 3.0

**Steps To Reproduce**
1. On a Raspberry Pi Bullseye (or any I tried), use the default operating system. Run sdrpp. The CPU load will be high.
2. On a Raspberry Pi Bullseye, edit /boot/config.txt and specify dtoverlay=vc4-fkms-v3d as show above. CPU load will be much lower.

Can SDR++ be made to be more compatible with the newer OpenGL on the Raspberry Pi?
Debian Bookworm comes with a driver that reports
OpenGL version string: 4.5 (Compatibility Profile) Mesa 22.2.0-rc1
server glx version string: 1.4

----Steve

AlexandreRouma, 2022-08-16:
You've clearly got an issue with your graphics driver, nothing I can do.

> Can SDR++ be made to be more compatible with the newer OpenGL on the Raspberry Pi?

It has nothing to do with the OpenGL version... Asking the loader for newer versions won't magically increase performance.
You're running software rendered instead of using the real 3D acceleration driver. SDR++ has been well tested on the pi4 with the RaspioOS image and has no such issue, so the distro you're using is almost certainly the problem.

closed 2022-08-16 by AlexandreRouma

srs4511351, 2022-08-16:
This problem also exists in the standard Raspberry Pi images. It would be good if you try it. You must have tried a very old image. 
Time moves on and drivers are updated. 
I build systems with many SDR applications and none of them have problems with the graphics driver. It’s unique to SDR++.

As indicated in my message, the old outdated driver works, but two different and more current drivers do not. Not the fault of the driver. 
Are there packages that can influence this?

it’s not about new drivers increasing performance, but about SDR++ being compatible with current OSes. It’s apparent that SDR++ is doing the rendering when CPU use is high. 

——Steve

AlexandreRouma, 2022-08-16:
> It would be good if you try it. You must have tried a very old image. Time moves on and drivers are updated.

I run almost the latest official RaspiOS Bullseye image on my pi and have no such issue.
In addition many users report using it just fine with the latest official image.

> As indicated in my message, the old outdated driver works, but two different and more current drivers do not. Not the fault of the driver. Are there packages that can influence this?

Well that should prove the "more recent" driver is broken... It's likely not installed or configured properly. I've never ever had to mess with drivers on the official raspberry pi images. Especially not graphics drivers...

> but about SDR++ being compatible with current OSes.

Using OpenGL 3.1 changes absolutely NOTHING in terms of compatibility with new operating systems. When a driver supports an OpenGL version, it automatically supports all the versions under it. All my PCs run rolling release distros, I always have the latest drivers and software, and I obviously daily drive SDR++ on those PCs, I know when something is compatible or not.

> It’s apparent that SDR++ is doing the rendering when CPU use is high.

SDR++ doesn't chose to do the rendering in software, it loads whatever OpenGL driver is installed and uses that. If your graphics driver isn't installed or setup properly it'll just use Mesa instead and render on the CPU.

srs4511351, 2022-08-16:
My graphic driver comes on the official operating system image from the Raspberry Pi OS developer. I did not install that. 
It only works when I do mess with it. 
The fact that the current one fails only with SDR++ and the newer one on the bookworm image does the same, points away from the drivers.

You added support for OpenGL 2.1 to fix the Raspberry Pi in issue #260. It shows that you can fix things. Issue #178 also discusses Raspberry Pi issues. 

Are there other factors or packages that can affect this? What configuration changes can affect this?

I doubt that other users of these images have unexplained successes, as this is discussed in other issues. They may even think that the high CPU load is normal.

Again, my 6 other SDR applications do not have problems with my OS. Additionally, my gui utilities and web browsers work well. What is different about SDR++?

Thank you for redoing the filters. They are much better now. 
looking forward to the notch filter and playing with the noise blanker. 

——Steve

srs4511351, 2022-08-17:
I tried my Raspberry Pi OS installation.
I remembered having to change the video driver, but it was for the audio.
With this driver, Pulse audio is choppy, but ALSA is clear.
The video does work with the original driver that comes with Raspberry Pi OS, so I must apologize for being incorrect about that. My memory failed me.
I believe that this driver works because is is still the old OpenGL version:
OpenGL version string: 2.1 Mesa 22.2.0-rc1

The system that I was working with was DietPi Bullseye and Bookworm. They are normally quite a bit ahead of Raspberry Pi by testing and releasing the latest Debian versions right after they are official, etc.
It still goes back to the drivers, but again, they work with everything I have installed except SDR++.
That problem remains. I do not believe that the new drivers are defective, but newer. They surely will be supplied with operating system images in the future. Maybe you can help them get it right?

When you made SDR++ work with the Raspberry Pi OpenGL 2.1 in  issue #260, was it fixed at that version?

----Steve

AlexandreRouma, 2022-08-17:
> You added support for OpenGL 2.1 to fix the Raspberry Pi in issue https://github.com/AlexandreRouma/SDRPlusPlus/issues/260. It shows that you can fix things. Issue https://github.com/AlexandreRouma/SDRPlusPlus/issues/178 also discusses Raspberry Pi issues.

It was not added to fix raspberry pi issues, it was to run properly on slightly older PCs. It only tries OpenGL 2.1 if it wasn't able to start up with OpenGL 3.0 or OpenGL ES 3.1.

> Again, my 6 other SDR applications do not have problems with my OS. Additionally, my gui utilities and web browsers work well. What is different about SDR++?

Your other SDR softwares dont use OpenGL...
using OpenGL for SDR software is the design choice that made SDR++ what it is.
No other software reaches the performance and cross platform support that SDR++ has.

> I remembered having to change the video driver, but it was for the audio.

There is no driver to switch for audio.... use the normal RTAudio fix: #112 

> OpenGL version string: 2.1 Mesa 22.2.0-rc1

You realize that's the latest OpenGL version the Pi4's GPU supports right?...
The alternative is OpenGL **ES** 3.1 and that's why SDR++ uses in priority.
No driver update will change that, what you think is a driver update is just installing OpenGL 3.x **software** support.
You won't magically change the hardware in the GPU...

Just install the latest official RaspiOS Bullseye image, apply the RTAudio fix and install SDR++
everyone runs it just fine and messing with drivers or alternative distros will only make things worse.

srs4511351, 2022-08-17:
The HDMI driver switch back to vc4-fkms-v3d did indeed fix some audio problems. I am using HDMI audio, as the built-in analog is poor quality. It allowed access to both hardware devices, Analog and HDMI. It also cures other flakey Pulsaudio problems. It works with SDR++ and improves audio support on it and my other SDR applications.
The RtAudio fix is a compromise, but does work for the purpose in which it was intended.

I was not aware that OpenGL 2.1 was the latest. The RPI OpenGL specification is very misleading. The official specification is OpenGL ES 3.1, Vulkan 1.0
If that means OpenGL 2.1, then they should say it!

On Raspberry Pi OS and DietPi, everything works with the alternate  vc4-fkms-v3d driver, no problems with it. It works better than the Raspberry Pi default dtoverlay=vc4-kms-v3d driver, including SDR++. All of the devices become available in CubicSDR.
Thanks for helping to understand the issues with the OpenGL 3.1 or higher. I should have a discussion with DietPi about using them.

----Steve