AlexandreRouma/SDRPlusPlus #851: Error while building for MacBook Air M1 12.5.1 (Monterey)

lawendel, 2022-08-23:
I'm following compiling notes provided by Techprepper 
[here](https://www.thetechprepper.com/sdr/installing-sdr-on-macos-monterey) 
 to try and build on my MacBook Air M1 w/ Monterey 12.5.1

Besides a few issues with dependencies installed with Homebrew (list of dependencies provided by Techprepper doesn't overlap with the list provided in the building instructions in this repository) and a long series of warnings mainly connected with deprecation issues  - final `make -j$(sysctl -n hw.ncpu)` command produces a fatal error:

`/Users/lawendel/SDRPlusPlus/source_modules/sdrplay_source/src/main.cpp:9:10: fatal error: 'sdrplay_api.h' file not found
#include <sdrplay_api.h>`

I'm not sure what to do, where do I find this .h file and where should I save it? Would that be enough to have a usable binary?

Thanks

AlexandreRouma, 2022-08-23:
the macos instructions are out of date but that issue is b ecause you left the SDRplay source build enabled without installing the SDRplay API.

lawendel, 2022-08-23:
So it's just a matter of configuring 

`-DOPT_BUILD_SDRPLAY_SOURCE=OFF`?

Sorry for being so step-by-step minded but I'm really a newby...

AlexandreRouma, 2022-08-23:
yes, though other stuff is require to actually do things correctly. haven't currently got the time to explain any of this unfrotunately

lawendel, 2022-08-23:
The alternative is of course installing the API! There's an M1 beta on SDRplay's website. Thank you!

lawendel, 2022-08-23:
It turned out that the script for that SDRPlay API failed to install too (I suspect something related to environmental changes after Homebrew was installed). OTH, once option for SDRPlay module was turned OFF - and a huge salvo of "warning" messages during compile with make - I now have an M1 binary (RDS capable) I can launch from the Terminal! Also, CPU load is acceptable. 

I am super happy, for it must be my first significant build without Homebrew or Mac Ports. You made my day. Good luck with your uni tests.

refplusten, 2022-09-05:
lawendel - I think your level of newbie is much different than my level of newbie...  I have a Mac M1 pro running Monterey 12.5.1 and I cannot get this to load.  I followed instructions from Techprepper that you linked above.  I don't understand what you did to get this software to work.  Do you mind helping a fella out?  :-)

lawendel, 2022-09-05:
Matter is... I'm not sure exactly what, either! In my case, the reason for the single compromising error during compilatio was related to a driver for SDRplay devices which cmake was supposed to find. Since it has proved impossible to install the required SDRplay API file (in spite of a very helpful technical support on their side... But I believe a new installer for M1 is on the way), I simply turned OFF the device option for "BUILD_SDRPLAY_SOURCE" in cmake list you can find in: 
```
$ mkdir build && cd build
$ cmake .. \
```
(it's the last -DOPT line) I believe it's there because SDR++ supports SDRplay and other devices as source via SoapySDR. 

And that's all, after that cmake kept on generating minor errors but the program is usable all the same. I hope Alexandre's exams went fine and he'll find some time to update those MacOS build instructions. 

Anyway, refplusten, SDR++'s binary you can find in the "nightly build" section (go for 1.0.4 release, and pull down [sdrpp_macos_amd64.pkg] from the assets list) runs fine under Rosetta, without features like RDS decoding though. Hope this helps. 

Andy

AlexandreRouma, 2022-09-05:
Correction:

1) Do not edit the cmakelists files manually, use `-DBUILD_SDRPLAY_SOURCE=OFF`
2) Do not use the 1.0.4 pkg, it's old and broken and is NOT the nightly. The nightly is https://www.sdrpp.org/nightly

I can't update the MacOS instructions yet because I don't have a mac to even test anything.

Terminator0007, 2022-09-05:
SDRPlay API can install via MacPorts or Homebrew without any problems and all SDRPlay devices working correctly with M1 and Intel CPU and OS Monterey . I have Mac mini with M1 and Macbook with Intel m7. Both are working with SDRPlay devices even with chinese clone of RSP1.

lawendel, 2022-09-05:
Yes, but personally, I'm a bit confused by the x86/arm64 duality. On my M1 machine I have both SDR++ x86 under Rosetta and now my M1 build. I'm not sure if the latter would work with a "universal" API. That's why I have downloaded the beta installer of the SDRplay M1 API from their site. I have yet to test both with my RSPA1, so far only Airspy and RTL-SDR hardware were successfully used. What do you think?

lawendel, 2022-09-05:
> Correction:
>
> 1. Do not edit the cmakelists files manually, use `-DBUILD_SDRPLAY_SOURCE=OFF`
> 2. Do not use the 1.0.4 pkg, it's old and broken and is NOT the nightly. The nightly is [...]

Yes, sorry for the wrong URL, I forgot about the correct link from sdrpp.org.

refplusten, 2022-09-05:
Awesome!
By changing  -DBUILD_SDRPLAY_SOURCE=OFF and then running the nightly, it seems to work!!!
Thanks so much!

AlexandreRouma, 2022-09-30:
Since the original issue of the sdrplay library being missing is resolved, I'll close this issue.

closed 2022-09-30 by AlexandreRouma