AlexandreRouma/SDRPlusPlus #857: RSP_TSP stream - LSB = USB?

betzburger, 2022-08-31:
Good evening.
My setup: I have a Raspberry Pi and a RSPdx connected to it.
When I do a RSP_TSP stream to SDRPP on my Mac M1, let's say on 40 m, then all ham's voice is in USB but should be in LSB.
This issue is not the case when I use other software. It's really strange. My fault or an issue?

AlexandreRouma, 2022-08-31:
I don't have a M1 mac to test so I can't say much but it works fine on PC.
I suggest using SDR++ server instead of RSP-TCP.

betzburger, 2022-08-31:
Wasn't aware of SDR++ server. Just checked it out in a Youtube video. Looks VERY promising. I will try this as it seems to be much better than RSP-TCP. Thanks for the advice.

betzburger, 2022-09-01:
Besides the SDR++ server, which I will use from now on, 
I compiled a new version of SDR++ for Apple Silicon this morning and I also checked the Windows version on another machine with the RSP_TCP stream.

* The Windows version is fine.
* The Mac version is wrong. 

See the example screenshot? You can see that the modulation is wrong. It is USB but should be LSB. 

[image]

But this can't hold me back to to mention that you have done a fantastic and outstanding work with SDR++!

AlexandreRouma, 2022-10-03:
As I said, can't test it, and I won't be supporting RSP_TCP since it's a modified version of RTL_TCP. Supporting it in the RTL module would requires modifying the code in bad ways and supporting it in a separate module would yield quite a bit of code duplication. RSP-TCP was only as a hack to get it working with software that doesn't support the RSP.

closed 2022-10-03 by AlexandreRouma