AlexandreRouma/SDRPlusPlus #861: How to View Wider Portion of Spectrum?

gbeyer3, 2022-09-04:
Hello again, Alexandre, fantaaaastic SDR app you have developed!  Thank you so much!

Can you tell us how we can view a larger portion of the spectrum?  Possibly a command line switch?  As is at present, the FFT (I think that is what the top portion is called) width is 250 khz at maximum zoom out, even viewed full screen.  Thus, one must manually scroll back and forth to see a full band.  It would be great to see the entire 2M, Airband, and SWL Broadcast bands, etc. at one time. 

I understand a scanner feature is in the works, but in the meantime, seeing a much larger segment of the spectrum would be a big help in visual "scanning" what signals are available to listen to. 

Thanks again,
Greg

AlexandreRouma, 2022-09-04:
I assume you're relatively new to SDR, in which case i'd recommend reading up on the different settings of an SDR,.
You need to set the sample rate:
[image]

gbeyer3, 2022-09-04:
Yes, I'm fairly new to SDR.  I've seen that setting, but didn't really know what it did.  Thanks!

closed 2022-09-04 by gbeyer3