AlexandreRouma/SDRPlusPlus #865: Installing only sdrpp server

Terminator0007, 2022-09-07:
**Feature description**
Is it possible to install only sdrpp server for device ( or more sdr devices) without software panel to receiving and control ?

AlexandreRouma, 2022-09-07:
SDR++ Server is a mode of SDR++, it can't be separated. Moreover there is literally no reason to do it. It doesn't initialize the graphics system if in server mode.

closed 2022-09-07 by AlexandreRouma

Terminator0007, 2022-09-07:
Ok. It is clear, thanks.

unixpunk, 2024-03-30:
Hi @AlexandreRouma - Thanks for you work on this, excellent software!  I have this same question, with the reason being to avoid needing to install all of the X / graphics-related dependencies.  Any chance this is possible now given the time that has passed?  Use-case is a headless raspberry pi running sdrpp-server only as a remote sdr to satdump.  Thanks either way!

AlexandreRouma, 2024-03-30:
It's not possible.

KubaPro010, 2024-04-09:
it is possible, you just have to delete a lot of code, here i have a version of sdrpp which doesnt have any graphical stuff (unless imgui code, i was too lazy to remove that) and on my server a `make -j4` took 1 min and 20 secs, but now its doing only MiriSDR and right now i have no plans of adding more sources, anyways, this is the project: https://github.com/KubaPro010/SDRMinusMinus crappy but works as a server

AlexandreRouma, 2024-04-09:
what you don't realize is the entire point of SDR++ server is to not have two separate code bases to maintain.

KubaPro010, 2024-04-09:
.in polish you'd say "mówi sie trudno" (ask chatgpt what it means, google translate translates it literally) :)

unixpunk, 2024-04-10:
> it is possible, you just have to delete a lot of code, here i have a version of sdrpp which doesnt have any graphical stuff (unless imgui code, i was too lazy to remove that) and on my server a [...]

Thanks!  This gives me a reason to use my RSP1a more!

KubaPro010, 2024-04-14:
i've just removed imgui (not completly as the server had to send a gui in text which did use some imgui attributes) and now i saw that the file dropped from 1.5 megabytes with im gui to 485 kilobytes of the core module, the compilation speed isn't that much faster but the file size is better and imgui could have had some opengl imports, so no opengl required now