AlexandreRouma/SDRPlusPlus #871: Display squelch level on frequency display

b4ux1t3, 2022-09-16:
**Feature description**
One of the killer features (as in, the only feature I actually like) of the SDR software I use on Android is a visual display of the squelch level on the frequency graph. It also acts as the squelch level control, since there isn't much room on the screen for extra controls.

I would love to see this feature in SDR++, and I think it could be a standout feature, as I don't see it in any of the other (free) SDR software I've tried. It would make using SDR++ on a tablet just that much nicer.

AlexandreRouma, 2022-09-16:
There is no clean way to implement this. The squelch level has no relationship to the FFT. It's based on IF level which varies depending on quite a few things (IF NR, bandwidth, baseband samplerate, etc), It's just much more efficient and clean to do things that way, that's why you see no other software doing it based on the FFT level...

closed 2022-09-16 by AlexandreRouma

b4ux1t3, 2022-10-07:
Fair enough! I'm still unclear on how certain parts of the software fit together (still very new to radio). I figured it was just a matter of convenience in the mobile interface, but I didn't know how closely (or loosely) these factors were coupled.