AlexandreRouma/SDRPlusPlus #876: Record function not available on Linux

delboy85341, 2022-09-25:
I have a Dell Optiplex 7010.   /proc/cpuinfo says its an Intel(R) Core(TM) i5-3570 CPU @ 3.40GHz
It has 32GB RAM in it.  The OS is Kubuntu 22.04.1

$ lspci | grep VGA
01:00.0 VGA compatible controller: NVIDIA Corporation GK208B [GeForce GT 710] (rev a1)

SDRPlay RSP1a

I installed the latest nightly build for Ubuntu 22.04.  When I launch SDR ++ it says at the top: v1.1.0(Built at 13:26:41 Sep 5 2022

The record option doesn't work.  I click the record button and nothing happens and nothing gets recorded.  I also tried recording from my machine when it's booted into windows 10 and recording works.   I downloaded the zip file for windows at https://github.com/AlexandreRouma/SDRPlusPlus/releases/download/1.0.4/sdrpp_windows_x64.zip.  I also downloaded the source and compiled it in Kubuntu and the record option doesn't work either.  If you have any questions, let me know. thanks.

AlexandreRouma, 2022-09-26:
Firstly that version is quite out of date. Get the nightly:
https://www.sdrpp.org/nightly.
Secondly, this sounds to me like user error. Probably trying to record
somewhere that doesnt exist or without write permission. I suggest you post
both complete logs (run sdr++ in a terminal) as well as a screenshot of the
recorder menu.

delboy85341, 2022-09-26:
You're right. It was user error.  I corrected the problem and it works now.  .