AlexandreRouma/SDRPlusPlus #877: SDR++ on android tv

Mikeload66, 2022-09-26:
Hi there ,
At the moment I'm using sdr touch on my android tv box
But it has limited features , I'm needing bias tee switching so I can select another antenna , I've installed sdr++ apk file successfully but when I press play on the screen the program switches back to the homescreen.
Tv box is running android 11, any advice will much appreciated. 

Regards
Mike

AlexandreRouma, 2022-09-26:
You haven't given any useful information to debug the issue as asked for in the issue template...
What hardware? what specs? what SDR? Which SDR++ version? Nightly or first prerelease? Does the device show up in the list? Screenshots? Error message? etc.

AlexandreRouma, 2022-10-10:
closing as stalled, reopen when/if info asked for is provided

closed 2022-10-10 by AlexandreRouma