AlexandreRouma/SDRPlusPlus #878: Audio Error?

shadyslim2018, 2022-09-29:
Motherboard - X570 AORUS MASTER -CF 
Kernel: 5.19.7-1-MANJARO 
DE: GNOME 42.4 
CPU: AMD Ryzen 9 5950X (32) @ 3.400GHz 
GPU: AMD ATI Radeon RX 6700 XT
Memory: 32020MiB 
SDR : HackRF One Local

SDR++ v1.1.0, I tried from aur and built from scratch but same issue.

I've removed the modules I don't need.
``
[2022-09-29 09:06:29.230] [info] SDR++ v1.1.0
[2022-09-29 09:06:29.230] [info] Loading config
[2022-09-29 09:06:29.337] [info] Using OpenGL 3.0
[2022-09-29 09:06:29.376] [info] Loading icons
[2022-09-29 09:06:29.381] [info] Loading band plans
[2022-09-29 09:06:29.383] [info] Loading band plans color table
[2022-09-29 09:06:29.391] [info] Loading modules
[2022-09-29 09:06:29.391] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-09-29 09:06:29.393] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-09-29 09:06:29.394] [warning] ConfigManager locked, waiting...
[2022-09-29 09:06:29.394] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-09-29 09:06:29.395] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-09-29 09:06:29.396] [info] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[2022-09-29 09:06:29.396] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-09-29 09:06:29.397] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-09-29 09:06:29.398] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-09-29 09:06:29.399] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-09-29 09:06:29.400] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-09-29 09:06:29.401] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-09-29 09:06:29.402] [info] Loading /usr/lib/sdrpp/plugins/scanner.so
[2022-09-29 09:06:29.403] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-09-29 09:06:29.408] [info] Initializing Airspy Source (airspy_source)
[2022-09-29 09:06:29.408] [error] Module 'airspy_source' doesn't exist
[2022-09-29 09:06:29.408] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-09-29 09:06:29.408] [error] Module 'airspyhf_source' doesn't exist
[2022-09-29 09:06:29.408] [info] Initializing Audio Sink (audio_sink)
[2022-09-29 09:06:29.409] [info] Initializing BladeRF Source (bladerf_source)
[2022-09-29 09:06:29.409] [error] Module 'bladerf_source' doesn't exist
[2022-09-29 09:06:29.409] [info] Initializing File Source (file_source)
[2022-09-29 09:06:29.409] [info] Initializing Frequency Manager (frequency_manager)
[2022-09-29 09:06:29.409] [info] Initializing HackRF Source (hackrf_source)
[2022-09-29 09:06:29.412] [info] Initializing LimeSDR Source (limesdr_source)
[2022-09-29 09:06:29.413] [error] Module 'limesdr_source' doesn't exist
[2022-09-29 09:06:29.413] [info] Initializing Network Sink (network_sink)
[2022-09-29 09:06:29.413] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-09-29 09:06:29.413] [error] Module 'plutosdr_source' doesn't exist
[2022-09-29 09:06:29.413] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-09-29 09:06:29.413] [error] Module 'rtl_sdr_source' doesn't exist
[2022-09-29 09:06:29.413] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-09-29 09:06:29.413] [error] Module 'rtl_tcp_source' doesn't exist
[2022-09-29 09:06:29.413] [info] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

zsh: segmentation fault (core dumped)  sdrpp
```

shadyslim2018, 2022-09-29:
This is the error when I install it from the aur package. The installation goes fine, the problem is when running sdrpp

```
[2022-09-29 09:44:38.549] [info] SDR++ v1.1.0
[2022-09-29 09:44:38.549] [info] Loading config
[2022-09-29 09:44:38.620] [info] Using OpenGL 3.0
[2022-09-29 09:44:38.658] [info] Loading icons
[2022-09-29 09:44:38.662] [info] Loading band plans
[2022-09-29 09:44:38.664] [info] Loading band plans color table
[2022-09-29 09:44:38.666] [info] Loading modules
[2022-09-29 09:44:38.666] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-09-29 09:44:38.667] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2022-09-29 09:44:38.668] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2022-09-29 09:44:38.668] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2022-09-29 09:44:38.668] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-09-29 09:44:38.669] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-09-29 09:44:38.669] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2022-09-29 09:44:38.670] [info] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[2022-09-29 09:44:38.670] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2022-09-29 09:44:38.671] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-09-29 09:44:38.671] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-09-29 09:44:38.672] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-09-29 09:44:38.672] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2022-09-29 09:44:38.673] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2022-09-29 09:44:38.675] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2022-09-29 09:44:38.676] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2022-09-29 09:44:38.676] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2022-09-29 09:44:38.677] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-09-29 09:44:38.678] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-09-29 09:44:38.678] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2022-09-29 09:44:38.679] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2022-09-29 09:44:38.679] [info] Loading /usr/lib/sdrpp/plugins/scanner.so
[2022-09-29 09:44:38.680] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2022-09-29 09:44:38.682] [info] Initializing Airspy Source (airspy_source)
[2022-09-29 09:44:38.685] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-09-29 09:44:38.688] [info] Initializing Audio Sink (audio_sink)
[2022-09-29 09:44:38.688] [info] Initializing BladeRF Source (bladerf_source)
[2022-09-29 09:44:38.691] [error] Could not list devices -7
[2022-09-29 09:44:38.691] [info] Initializing File Source (file_source)
[2022-09-29 09:44:38.691] [info] Initializing Frequency Manager (frequency_manager)
[2022-09-29 09:44:38.692] [info] Initializing HackRF Source (hackrf_source)
[2022-09-29 09:44:38.695] [info] Initializing LimeSDR Source (limesdr_source)
[2022-09-29 09:44:38.701] [info] Initializing Network Sink (network_sink)
[2022-09-29 09:44:38.701] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-09-29 09:44:38.701] [info] Initializing RFspace Source (rfspace_source)
[2022-09-29 09:44:38.701] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-09-29 09:44:38.704] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-09-29 09:44:38.704] [info] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

RtApiJack::getDeviceInfo: error determining Jack input/output channels!

/usr/include/c++/12.2.0/bits/stl_vector.h:1123: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = RtAudio::DeviceInfo; _Alloc = std::allocator<RtAudio::DeviceInfo>; reference = RtAudio::DeviceInfo&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
zsh: IOT instruction (core dumped)  sdrpp

```

AlexandreRouma, 2022-09-29:
LibRtAudio bug, follow work around instructions: https://github.com/AlexandreRouma/SDRPlusPlus/discussions/112

shadyslim2018, 2022-09-29:
> LibRtAudio bug, follow work around instructions: #112

Excellent, thank you, it worked.

closed 2022-09-29 by AlexandreRouma