AlexandreRouma/SDRPlusPlus #883: Control low and high cut of a filter

ja-Bertolin, 2022-10-08:
In relation to filters,

1.- Is there any way to control Audio Filters?
2.- if so,  is there ay way to control the low and high cut frequency? For Wefax sometimes it is needed to reduce adjacent noise

AlexandreRouma, 2022-10-08:
1) There are no audio filter in SSB. VFO does all the work.
2) No. And it would absolutely not help for wefax.

bobhwasatch, 2022-10-13:
FWIW my SDR computer runs Debian Linux with XFCE. The default setup uses PulseAudio and I've had success using PulseEffects to filter the audio from SDR++. This allows for things like inserting a high-pass filter to remove the squelch tone from 2-meter repeaters, using the equalizer for reducing noise in SSB, etc. There's a lot of plugins for PulseEffects, as I said I make the most use of the filter and equalizer, but the limiter and compressor are also helpful for some things.

It is a little fiddly to get set up (have to start SDR++ first, sometimes have to fix the routing with pavucontrol), but some of that might be my setup and the fact that I'm running Bookworm/Testing.

darksidelemm, 2023-03-05:
I'll add a +1 for a high-pass cutoff setting for the SSB demodulator.

Having used SDR++ for a long time to listen around on HF, being able to add in a bit of high-pass cutoff (remembering on most SSB rigs, 300 Hz is the default for this) helps a lot with dealing with adjacent-channel interference, people tuning up on or near the VFO frequency (my room shakes due to the bass produced when this happens...), and just getting rid of other annoying noise in general. The low-pass portion of the SSB filter is already adjustable (through the adjusting of the SSB bandwidth), high-pass adjustment would be a much appreciated addition. 

I agree that it won't help with data decoding (this should be handled by the modem), but for general SSB listening it *is* important.

I can also echo the comments about hearing CTCSS tones when demodulating FM. This is another situation where a high-pass filter is useful.

craigerl, 2023-03-15:
Set the audio sink in sdr++ to "network" then run,

nc -l -u 7355 |  play --buffer 100 -t raw -b 16 -e signed -r 48k - -r 16k sinc 0.3k  -n 3000

(apt-get install sox if "play" not found)

good bye PL tone!

darksidelemm, 2023-03-15:
This is a nice workaround (if you are running linux/OSX), but not really a solution...

craigerl, 2023-03-15:
Yah, agreed, it's not a solution.  gqrx isn't interested in filtering pl tones either.   I couldn't listen to repeaters on sdrpp until i started using that filter.

AlexandreRouma, 2023-03-15:
The reason it's a feature to be added and not done by default is the demodulator is meant to also receive digital signals. High-pass filtering would make this impossible.

darksidelemm, 2023-03-15:
Sure, I wouldn't necessarily expect it to be on by default, but having it available makes using SDR++ as a listening tool much more enjoyable... 
I'm guessing this would end up being another setting box to set the high-pass cutoff, and some way of dragging the filter envelope on the waterfall (kind of like how the openwebrx / kiwisdr interfaces do it).

AlexandreRouma, 2023-03-25:
High pass on **FM** was just added to deal with the PL tones.

StvLa, 2023-04-10:
Nice. What is the cutoff frequency of the high-pass filter ?

AlexandreRouma, 2023-04-10:
300Hz

darksidelemm, 2023-05-06:
Can we get this on SSB as well please?