AlexandreRouma/SDRPlusPlus #884: Only 48Khz option for sample rate on SoapyAudio for I/Q based interfaces/radios.

fyoory, 2022-10-08:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: AMD Ryzen 7 5700G
- RAM: 32G
- GPU: Nvidia GTX 2080 
- SDR: Local I/Q Audio via SoapySDR

**Software**
- Operating System: Manjaro 22.0.0 (AUR pkgbuild)
- SDR++: 1.1.1 from AUR build sdrpp-git build, built today.
**Bug Description**
SoapySDR driver works however I can only select 48Khz bandwidth.  My soundcard is an ASUS XONAR SE 5.1 Channel 192kHz/24-bit (https://www.amazon.com/ASUS-XONAR-SE-Channel-Compatibility/dp/B07HCX1NY9)

**Steps To Reproduce**
1. Start sdr++
2. Set Source to SoapySDR
3. Set the appropriate sound card name below that
4. Drop down for samplerate only has 48Khz,

**Screenshots**
[image]

CubicSDR seems to read this fine with SoapyAudio
[image]

Both seem to link to soapysdr 0.8.1 library.

**Additional info**
Not sure if this is related to SoapyAudio/SoapySDR or not.  However since CubicSDR allows for it as shown above, it makes me think it is app specific to sdr++.   If it helps I use the Xonar Soundcard with my Elecraft KX3 ham radio which provides I/Q Audio out.    Separated issue that I dont think 192K works at all in linux, or doesnt at least in CubicSDR.  96k and the lowers do tho.  96K would be livable, as I normally use that since the soundcard itself gets kind of noisey at outside of the 96k spread when you use 192k.

AlexandreRouma, 2022-10-08:
SDR++ shows every single samplerate that SoapySDR provides. On every hardware I have and software (pulseaudio), all samplerates are availalble.

If these options are not listed, it's an issue with soapysdr (related to your exact audio setup) and an issue should be opened with them instead. Cubic might be overriding options given by soapy which is bad (what's the point of having an abstraction library if you have to override stuff for specific sources...).

PS: Native audio support will be added in the future.
Also, you're running an over 1 year old version. The [nightly](https://www.sdrpp.org/nightly) is almost release worthy so I'd suggest upgrading to that instead.

closed 2022-10-08 by AlexandreRouma

fyoory, 2022-10-09:
commit c95c7b18af2fa58fb5f977e0c55fecdafd83f52e (HEAD -> master, origin/master, origin/HEAD)
Author: AlexandreRouma <alexandre.rouma@gmail.com>
Date:   Tue Sep 27 15:43:33 2022 +0200

    added invert iq option 

I checked the PKGBUILD work dir and this is the latest commit.  So I dont *think* this is over a year old, at least that is what 'git log' says.  It is at Arch/Manjaro AUR sdrpp-git build, but I may have made some mistakes typing up this yesterday in Haste.

Is there some easy way for me to confirm it is soapy?

fyoory, 2022-10-09:
Also, I am highly interested in the direct audio version.  Would love it, I am not very satistifed historically speaking with Soapy.    Pulse, pipewire or Jack2 would be more ideal than soapy.

AlexandreRouma, 2022-10-09:
> So I dont think this is over a year old,

I can see your version is a year old from the version text in your screenshot. 1.0.4 was from october 2021.
The `sdrpp` package isn't official, some random person make it and apparently only chose main releases (Which I honestly don't feel like doing with how fast the development cycle for SDR++ is) instead of the nightlies.
I should probably reclaim it when I get more time. The `sdrpp-git` package is the official one and still says 1.0.4 because the arch PC I used to create it is out of service at the moment and I can't update the pkgbuild.

> Is there some easy way for me to confirm it is soapy?

There is nothing to confirm. If it's not listed, soapy doesn't list it as supported. The code is as clear as it gets about it: https://github.com/AlexandreRouma/SDRPlusPlus/blob/c95c7b18af2fa58fb5f977e0c55fecdafd83f52e/source_modules/soapy_source/src/main.cpp#L196-L209

fyoory, 2022-10-10:
Ok thanks.    Yea the pkgver I think probably is why it  'rendered' it as 1.0.4.  Or just as easily, I could have pasted the wrong screen shot too in fairness.  I tried both 1.0.4 and the below with the same results.  So i dont think nightly would fix it as you've indicated the issue is soapy.  

Thank you for your time on this.

``` $ cat PKGBUILD
# Maintainer: Alexandre Rouma <whatsthetgeek@gmail.com>
pkgname=sdrpp-git
pkgver=1.0.4
pkgrel=5
epoch=
pkgdesc="The Bloat-free SDR Receiver"
arch=('any')
url="https://github.com/AlexandreRouma/SDRPlusPlus"
license=('GPL')
groups=()
depends=("fftw" "glfw" "glew" "libvolk" "rtaudio")
makedepends=("gcc" "make" "cmake" "soapysdr" "libiio" "libad9361" "rtl-sdr" "airspy" "airspyhf-git" "hackrf" "libsdrplay" "bladerf" "limesuite")
checkdepends=()
optdepends=("soapysdr" "libiio" "libad9361" "rtl-sdr" "airspy" "airspyhf-git" "hackrf" "libsdrplay" "bladerf" "limesuite")
provides=()
conflicts=()
replaces=()
backup=()
options=()
install=
changelog=
source=('git+https://github.com/AlexandreRouma/SDRPlusPlus')
noextract=()
md5sums=('SKIP')
validpgpkeys=()

build() {
        cd "SDRPlusPlus"
        mkdir build
        cd build
        cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON -DOPT_BUILD_BLADERF_SOURCE=ON -DOPT_BUILD_LIMESDR_SOURCE=ON
        make $MAKEOPTS
}

package() {
        cd "SDRPlusPlus/build"
        make DESTDIR="$pkgdir/" install
}

pkgver() {
        cd "SDRPlusPlus"
        printf "r%s.%s" "$(git rev-list --count HEAD)" "$(git rev-parse --short HEAD)"
}```

AlexandreRouma, 2022-10-10:
the nightly indeed wouldn't fix anything related to the 48KHz thing but it's more efficient, has was better DSP, more features, etc.
Was just a general recommendation.