AlexandreRouma/SDRPlusPlus #885: Need some guidance on making a custom module

caballero03, 2022-10-09:
Hello,

I am planning to create a custom module for SDR++ that will react to frequency changes and automatically change a 10 band HF preselector and its bypass. I am running this on a Raspberry Pi 4 and using the GPIO pins to control the preselector bands.

My question is can someone direct me in how I can receive frequency changes in the GUI (or I suppose from remote, etc)?

Also, should this be a misc. module? Or maybe a decoder module?

Any clues would be super helpful. Thanks in advance.

AlexandreRouma, 2022-10-10:
There's a demo module you can use as a base: https://github.com/AlexandreRouma/SDRPlusPlus/tree/master/misc_modules/demo_module

>  how I can receive frequency changes in the GUI 

I assume you mean changed in the center freq of the SDR. On the master branch there is no way to get notified of SDR retuning, but on the `rigctl_client` branch which will be merged in a few weeks there is. just needs to register an even with the source manager.

> Also, should this be a misc. module? Or maybe a decoder module?

Doesn't matter, it's just for organisation and I don't currently acceppt module pull request anyway (due to code quality concerns) so feel free to do it how you want in your code.

caballero03, 2022-10-10:
Thanks for the advice. I think I may be able to get this to work now. Very helpful. Looking forward to the next release.

closed 2022-10-10 by caballero03