AlexandreRouma/SDRPlusPlus #887: M17 decoder improvements

sp5wwp, 2022-10-14:
**Feature description**
- Add a value entry field (or a drop-down list) to set the CAN number. An additional tick box can be used to allow for monitoring all CANs.
- Mute encrypted M17 traffic.
- Add a tick box to enable persistence of the last heard transmission (source, destination etc.).

TA1EE, 2022-10-25:
I've never been successful in monitoring the encrypted M17 traffic.

AlexandreRouma, 2022-10-25:
that's the entire point of it being encrypted...

sp5wwp, 2022-11-18:
Do you use hard or soft viterbi decoder for M17?

AlexandreRouma, 2022-11-18:
Hard and yes I know soft is what's supposed to be used but I don't have the time to change it since it requires a completely rewrite of the demod. It was written in 2 days, not really supposed to be the best.

k5jae, 2023-03-17:
Another improvement would be to support 1600 V/D of M17. MVoice supports it as an option and some folks are using it.

Paulo-D2000, 2023-05-24:
[FEATURE] My try at implementing the encryption mute feature for m17... It build on windows... should pass on the CI too (except ARM?)👍 
https://github.com/Paulo-D2000/SDRPlusPlus/commit/b4a6a3febb8f77aeebc55e3939d3ae38e6b683d9

TuxedoHam, 2023-11-12:
I see this enhancement has been around for a year now, as much as it is really nice to have the M17 decoder, but it would be really nice to have a BER output/display to be added. It be a great help for us that use SDR++ to tune in our Module17 modem boards