AlexandreRouma/SDRPlusPlus #892: Future request:  Radiosonde decoder plugin module

9w2www, 2022-10-18:
**Feature description**
Requesting to add Radiosonde decoder in plugin module please.

9w2www, 2022-10-18:
> There's already a third party plugin. Adding a first party one is not planned for now. https://github.com/dbdexter-dev/sdrpp_radiosonde

Sorry forgot to mention for sdr++ android.  can I add the 3rd party plugin?

AlexandreRouma, 2022-10-18:
Requires recompiling the app. If you are not familiar with android development I wouldn't attempt it.

9w2www, 2022-10-18:
> Requires recompiling the app. If you are not familiar with android development I wouldn't attempt it.

No i am not familiar and will not attempt anything.  Is that mean you will not include it in the future or have radiosonde plugin for sdr++ android?.  I am just this close to find the easiest app / tool to use for  mobile/ portable tracking radiosonde with my rtl sdr dongle. :(

AlexandreRouma, 2022-10-18:
I never include any third party plugins in official build. I can't possibly support code I didn't write.
Ask the author of the module to provide an APK if you want to have it on android.

9w2www, 2022-10-18:
> I never include any third party plugins in official build. I can't possibly support code I didn't write. Ask the author of the module to provide an APK if you want to have it on android.

I understand. Thank you for reply and explanation. no worry. i'll try asking the other party. :)

scottjpalmer, 2023-01-25:
I enlisted the help of a friend who got it working.
Download here https://github.com/jethrocarr/SDRPlusPlus/releases/download/nightly/sdrpp.apk
Video of it working here https://youtu.be/Ru0BRHeV7b0
It is using the latest SDR++ nightly as of today.