AlexandreRouma/SDRPlusPlus #893: [Feature Request] recording file name templates

thomasdouwes, 2022-10-20:
A text box or config file entry in the record module allowing you to enter a file name template, the current template could be the default as
`%(type)_%(frequency)_%(hour)-%(minute)-%(second)_%(day)-%(month)-%(year).wav`
or you could use a largest to smallest date with 
`%(type)_%(frequency)_%(year)-%(month)-%(day)_%(hour)-%(minute)-%(second).wav`
or something different like
`%(year)-%(month)-%(day)_%(hour)-%(minute)-%(second)_%(type)_%(frequency).wav`
or even subdirectories with
`%(year)-%(month)-%(day)/%(type)_%(frequency)_%(year)-%(month)-%(day)_%(hour)-%(minute)-%(second).wav`
this would be useful for organising recordings without manually sorting them
I have had a go at this and got it mostly working except for %(type) and subdirectory's, my code is awful but i could submit a pull request after tidying up my code

AlexandreRouma, 2022-10-20:
already planed but not gonna be as verbose, it has to fit in the side menu.

romanremus, 2022-12-01:
hello everyone......can you please compile sdr++ also for windows 7 /64 bit.....or please provide support in the source code for compiling in cygwin...........I have I tried to open it with windows 7 but I don't have SoapySDR.dll.....and in compilation with cygwin...I don't have certain dependencies and I can't install them in cygwin

thomasdouwes, 2023-01-03:
Closing because of name template in new recorder

closed 2023-01-03 by thomasdouwes

rs1729, 2023-03-27:
Is it possible to add radio demodulation type ("NFM", "WFM", "AM", ..., "RAW") to "Audio" recording filename?

AlexandreRouma, 2023-09-06:
For the record this was implemented