AlexandreRouma/SDRPlusPlus #895: Adalm Pluto distorted sound

loewal, 2022-10-21:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
iMac 24"

**Software**
Monterey

**Bug Description**
Sound, especially WFM sounds distorted heavily. Switching to Sdrplay RSP1 ands sound is good.

**Steps To Reproduce**
As described. Trying various settings: no change at all

loewal, 2022-10-21:
Switched to GQRX and also SDRangel. In both everything is fine.

AlexandreRouma, 2022-10-21:
Almost certainly user error and you haven't given any screenshot or audio sample that would actually allow me to check anything.

loewal, 2022-10-21:
Sorry Alexandre,

I will try to complete my issue:
- Macos Monterey 12.6 Intel
- downloaded v.1.1.0 built at Oct 19 2022.
[image]

https://user-images.githubusercontent.com/7289719/197213243-d4e543d2-05b3-4cdd-9193-9c1a03657086.mov

This file has extension mov instead of wav; please change it. I could not attach a wav-file.
Sorry for this!

[image]

loewal, 2023-01-06:
Addendum:
The Pluto works fine in GQRX, SDRAngel and "partly" under CubicSDR.
I installed SoapyPlutosdr and now I can see Pluto listed under available devices with 192.168.2.1
This works, but when I add a manual device with 192.168.2.1 as Pluto, the distortion is exactly the same as in SDR++

What is your conclusion? Don't you use Soapy for Pluto?

AlexandreRouma, 2024-01-22:
Now that I actually have a mac I can replicate this.

AlexandreRouma, 2024-01-22:
Unfortunately, this looks like a bug of IIO's network mode on MacOS.
Switching to USB mode without changing anything in the code results in perfect sound instantly...

So I think the only fix here is to finish proper USB device enumeration instead of relying on the networked mode.

AlexandreRouma, 2024-01-29:
Implemented.

closed 2024-01-29 by AlexandreRouma