AlexandreRouma/SDRPlusPlus #903: Memory or preset for scanner

Terminator0007, 2022-10-28:
**Scanner memory** 
Thanks so much Alexandre for very good scanner for me this is a feature number 1 in integrated in SDR++ but ... Always when user switch on sdrpp must always setup frequency ( start & stop ) , interval and signal level. Is it possible to memorize last settings and after restart or new work with sdrpp to stay with a last scanner settings ? Default settings are kind for FM transmission but I'm generally listen UHF or VHF stations and always must again setup scanner. I do not expect some presets or other solutions but it could be great to start sdrpp with last scanner settings same like for general VFO which after switching on back to the last band and frequency.. What you think about it ?

AlexandreRouma, 2022-10-28:
it's on purpose because it's beta and not ready for use. I don't want to deal with converting config formats late.

Terminator0007, 2022-10-28:
Ok, thanks but maybe in the future you do not forget about it. For this scanner feature is Gold Medal for you :-)