AlexandreRouma/SDRPlusPlus #927: SoapySDR device arguments/custom device string

LunaWuna, 2022-11-28:
I have looked around and cant find a way to use a specific soapysdr string, thanks.

romanremus, 2022-12-01:
hello everyone......can you please compile sdr++ also for windows 7 /64 bit.....or please provide support in the source code for compiling in cygwin...........I have I tried to open it with windows 7 but I don't have SoapySDR.dll.....and in compilation with cygwin...I don't have certain dependencies and I can't install them in cygwin

AlexandreRouma, 2022-12-26:
@romanremus don't post on other people's issues. Open your own.

nmaster2042, 2023-04-09:
@AlexandreRouma : I know Soapy_source will be deprecated but I think it can stay disabled by default.

Here is a soapy_source mod made by @ericek111 witch add extra settings dynamically.
see here: https://github.com/ericek111/SDRPlusPlus/blob/soapysettings/source_modules/soapy_source/src/main.cpp

It actually parses ArgInfo and each item on the UI according to the type of the setting value.

For SDRPlay Devices, only small glitch is there is a IQ Correction settings from soapy and on already in SDRPP.

[image]

But This add better control for devices that don't have native support.

AlexandreRouma, 2024-04-10:
Soapy support is now deprecated.

closed 2024-04-10 by AlexandreRouma