AlexandreRouma/SDRPlusPlus #941: rigctl server and new hamlib used by WSJT-X

frspin, 2022-12-20:
Version 2.5.4 of WSJT-X use a new command for setting frequency:
F VFOX [freq]

where X is A|B and so

but rigctl server in SDR++ is awaiting only a command as:

F [freq]

Regards

Franco Spinelli
IW2DHW

AlexandreRouma, 2022-12-20:
duplicate of  #573
please follow that issue instead

closed 2022-12-20 by AlexandreRouma

lwvmobile, 2022-12-29:
Just for clarity, which of these syntax is the 'correct' syntax? I currently have DSD-FME written to use the F [freq] approach, and that works just fine in SDR++ and GQRX. If this gets changed, then its going to break compatibility with one or the other, so I'll have to write additional lines to retry with the new syntax if the old syntax returns RPRT 1.

frspin, 2022-12-29:
Il 29/12/22 09:04, lwvmobile ha scritto:
> Just for clarity, which of these syntax is the 'correct' syntax? I
> currently have DSD-FME written to use the F [freq] approach, and that
> works just fine in SDR++ and GQRX. If this gets changed, [...]

As far as I know there is no "correct" syntax.

Many programs, also rigctl, send F [freq] but WSJT-X (current version, 
not beta) send F [vfo] [freq].
So the correct behavior for SDR++ is to test how many tokens are in 
command. 2 tokens mean F [freq], 3 mean F [vfo] [freq]

Regards

Franco Spinelli
IW2DHW

ve9qrp, 2023-03-11:
I ran "nc -l -p 4532" to sniff what additionally WSJT might request. I found that when pressing the 'Test CAT' button on the Radio Settings tab, it  called the following before timing out with an error:
\chk_vfo
\dump_state

In fact, it will start any session with checking these, and I couldn't get it to start a simple frequency request without doing so. The TLDR is that minimal compatibility with WSJT will require a valid response to these two Hamlib commands alongside any sort of frequency control.

ve9qrp, 2023-03-11:
The following pexpect script was written to keep WSJT-X happy in querying a hamlib port. It's based on sniffing the corresponding responses of gqrx, whose code of course works and might be a good starting point for developing the corresponding code in SDR++. It of course does not accurately report the state of SDR++. However, it does show just how extensive the WSJT-X querying is before it gets to simply asking to change the frequency!

`import pexpect
state="""0
2
1
0.000000 10000000000.000000 0x2ef -1 -1 0x1 0x0
0 0 0 0 0 0 0
0 0 0 0 0 0 0
0xef 1
0xef 0
0 0
0x82 500
0x82 200
0x82 2000
0x221 10000
0x221 5000
0x221 20000
0x0c 2700
0x0c 1400
0x0c 3900
0x40 160000
0x40 120000
0x40 200000
0 0
0
0
0
0
0
0
0
0
0x40000020
0x20
0
0\r"""
child = pexpect.spawn('nc -l -p 4532')
child.expect('\\\chk_vfo')
child.send('RPRT 1\r')
child.expect('\\\dump_state')
child.send(state)
child.expect('v')
child.send('VFOA\r')
child.expect('f')
child.send('27057100\r')
child.expect('V VFOB\r')
child.send('RPRT 1\r')
child.expect('s\r')
child.send('0\rVFOA\r')
child.expect('m\r')
child.send('USB\r2700\r')
child.interact()
`

AlexandreRouma, 2023-03-11:
I already know about all of that.
I just do not have time to work on that stuff at the moment.

ve9qrp, 2023-03-11:
I apologise. I didn't mean to set an agenda for you: you're already doing so much great stuff! I just thought I'd document the issues for anyone who is interested.