AlexandreRouma/SDRPlusPlus #942: Unable to compile for SDRPlay

f5bcb, 2022-12-22:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: 
- RAM:
- GPU: 
- SDR: (Remote or local? If remote, what protocol?)

**Software**
- Operating System: Name + Exact version (eg. Windows 10 x64, Ubuntu 22.04, MacOS 10.15)
- SDR++: Version + Build date (available either in the window title or in the credits menu which you can access by clicking on the SDR++ icon in the top right corner of the software).

**Bug Description**
A clear description of the bug.

**Steps To Reproduce**
1. ...
2. ...
3. ...

**Only If SDR++ fails to lauch or the SDR fails to start:**
Run SDR++ from a command line window with special parameters:
* On Windows, open a terminal and `cd` to SDR++'s directory and run `.\sdrpp.exe -c` (if running SDR++ version 1.0.4 or older, use `-s` instead, though you should probably update SDR++ instead...)
* On Linux: Open a terminal and run `sdrpp -c`
* On MacOS: Open a terminal and run `/path/to/the/SDR++.app/Contents/MacOS/sdrpp -c`
Then, post the **entire** logs from start to after the issue. **DOT NOT truncate to where you *think* the error is...**

**Screenshots**
Add any screenshot that is relevant to the bug (GUI error messages, strange behavior, graphics glitch, etc...).

**Additional info**
Add any other relevant information.

f5bcb, 2022-12-22:
Raspberry pi 4, OS Raspbian Bulleye
The result show that a file sdrplay_api.h is missing 👍 

/home/pi/SDRPlusPlus/source_modules/sdrplay_source/src/main.cpp:9:10: fatal error: sdrplay_api.h: No such file or directory
    9 | #include <sdrplay_api.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [source_modules/sdrplay_source/CMakeFiles/sdrplay_source.dir/build.make:82: source_modules/sdrplay_source/CMakeFiles/sdrplay_source.dir/src/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1746: source_modules/sdrplay_source/CMakeFiles/sdrplay_source.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
pi@raspberrypi:~ $ fatal error: sdrplay_api.h: No such file or directory
-bash: fatal: command not found
pi@raspberrypi:~ $

AlexandreRouma, 2022-12-23:
You didn't install the sdrplay API.
Follow the instructions on SDRplay's website to install it. (only the API, ***NOT*** the build tools with soapy and other software)

f5bcb, 2022-12-23:
Thanks a lot.
Should I use the 2.10 stable version or the 3.06 Beta version ? ( OS is Raspbian 64 Bullseye ) ?

AlexandreRouma, 2022-12-23:
The latest 3.x version. the exact minor version isn't important (hence `x`).

f5bcb, 2022-12-23:
OK but the 3.x version for 64bits is not available ( Error 404 ) on the SDRPlay web site.
https://www.sdrplay.com/software/SDRplay_RSP_API-ARM64-3.06.1.run

AlexandreRouma, 2022-12-23:
Pretty sure the 32bit version should still run.
If not, it's something you should talk to SDRplay about.
It's their proprietary driver, nobody except them can compile or distribute it.

f5bcb, 2022-12-23:
Ok I'll try the 32bit, if not, I'll reinstall a 32bits OS ans if still not, I'll stay on Openwebrx !! Hi
Thanks again

kmp56, 2022-12-23:
https://github.com/awandahl/Tinker_Board_2/blob/main/SDRplay_RSP_API-ARM64-3.07.1.run

f5bcb, 2022-12-25:
Hi,
The device is a genuine RSP1-A connected to a Raspberry PI4 runing Bullseye 64bits OS
Using 64bits API, compile is OK (with a lot of warnings). 
Server start and connection is OK but I get an error 
[2022-12-25 09:37:56.483] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
This file exits in this place:
pi@Sdrpp:/usr/lib/sdrpp/plugins $ ll
total 13928
-rw-r--r-- 1 root root  535640 24 déc.  10:33 airspyhf_source.so
-rw-r--r-- 1 root root  539776 24 déc.  10:33 airspy_source.so
-rw-r--r-- 1 root root  550024 24 déc.  10:45 audio_sink.so
-rw-r--r-- 1 root root 1049584 24 déc.  10:48 discord_integration.so
-rw-r--r-- 1 root root  706568 24 déc.  10:34 file_source.so
-rw-r--r-- 1 root root  737872 24 déc.  10:49 frequency_manager.so
-rw-r--r-- 1 root root  535104 24 déc.  10:35 hackrf_source.so
-rw-r--r-- 1 root root  618512 24 déc.  10:36 hermes_source.so
-rw-r--r-- 1 root root  596592 24 déc.  10:46 meteor_demodulator.so
-rw-r--r-- 1 root root  564792 24 déc.  10:46 network_sink.so
-rw-r--r-- 1 root root  526096 24 déc.  10:44 plutosdr_source.so
-rw-r--r-- 1 root root  920224 24 déc.  10:47 radio.so
-rw-r--r-- 1 root root  778048 24 déc.  10:50 recorder.so
-rw-r--r-- 1 root root  571448 24 déc.  10:39 rfspace_source.so
-rw-r--r-- 1 root root  545168 24 déc.  10:51 rigctl_server.so
-rw-r--r-- 1 root root  545832 24 déc.  10:39 rtl_sdr_source.so
-rw-r--r-- 1 root root  534344 24 déc.  10:40 rtl_tcp_source.so
-rw-r--r-- 1 root root  473720 24 déc.  10:51 scanner.so
-**rw-r--r-- 1 root root  564240 24 déc.  10:41 sdrplay_source.so**
-rw-r--r-- 1 root root  597808 24 déc.  10:38 sdrpp_server_source.so
-rw-r--r-- 1 root root  559984 24 déc.  10:42 soapy_source.so
-rw-r--r-- 1 root root  625392 24 déc.  10:43 spectran_http_source.so
-rw-r--r-- 1 root root  541360 24 déc.  10:44 spyserver_source.so
pi@Sdrpp:/usr/lib/sdrpp/plugins $

kmp56, 2022-12-25:
Before compiling SDR++, I modified the CMakeLists.txt file to suit my needs
Its main part now looks like this:

# Backends
option(OPT_BACKEND_GLFW "Use the GLFW backend" ON)
option(OPT_BACKEND_ANDROID "Use the Android backend" OFF)

# Compatibility Options
option(OPT_OVERRIDE_STD_FILESYSTEM "Use a local version of std::filesystem on systems that don't have it yet" OFF)

# Sources
option(OPT_BUILD_AIRSPY_SOURCE "Build Airspy Source Module (Dependencies: libairspy)" ON)
option(OPT_BUILD_AIRSPYHF_SOURCE "Build Airspy HF+ Source Module (Dependencies: libairspyhf)" OFF)
option(OPT_BUILD_BLADERF_SOURCE "Build BladeRF Source Module (Dependencies: libbladeRF)" OFF)
option(OPT_BUILD_FILE_SOURCE "Wav file source" OFF)
option(OPT_BUILD_HACKRF_SOURCE "Build HackRF Source Module (Dependencies: libhackrf)" ON)
option(OPT_BUILD_HERMES_SOURCE "Build Hermes Source Module (no dependencies required)" OFF)
option(OPT_BUILD_LIMESDR_SOURCE "Build LimeSDR Source Module (Dependencies: liblimesuite)" OFF)
option(OPT_BUILD_SDRPP_SERVER_SOURCE "Build SDR++ Server Source Module (no dependencies required)" ON)
option(OPT_BUILD_RFSPACE_SOURCE "Build RFspace Source Module (no dependencies required)" OFF)
option(OPT_BUILD_RTL_SDR_SOURCE "Build RTL-SDR Source Module (Dependencies: librtlsdr)" ON)
option(OPT_BUILD_RTL_TCP_SOURCE "Build RTL-TCP Source Module (no dependencies required)" ON)
option(OPT_BUILD_SDRPLAY_SOURCE "Build SDRplay Source Module (Dependencies: libsdrplay)" ON)
option(OPT_BUILD_SOAPY_SOURCE "Build SoapySDR Source Module (Dependencies: soapysdr)" ON)
option(OPT_BUILD_SPECTRAN_SOURCE "Build Spectran Source Module (Dependencies: Aaronia RTSA Suite)" OFF)
option(OPT_BUILD_SPECTRAN_HTTP_SOURCE "Build Spectran HTTP Source Module (no dependencies required)" OFF)
option(OPT_BUILD_SPYSERVER_SOURCE "Build SpyServer Source Module (no dependencies required)" ON)
option(OPT_BUILD_PLUTOSDR_SOURCE "Build PlutoSDR Source Module (Dependencies: libiio, libad9361)" OFF)
option(OPT_BUILD_USRP_SOURCE "Build USRP Source Module (libuhd)" OFF)

# Sinks
option(OPT_BUILD_ANDROID_AUDIO_SINK "Build Android Audio Sink Module (Dependencies: AAudio, only for android)" OFF)
option(OPT_BUILD_AUDIO_SINK "Build Audio Sink Module (Dependencies: rtaudio)" ON)
option(OPT_BUILD_PORTAUDIO_SINK "Build PortAudio Sink Module (Dependencies: portaudio)" OFF)
option(OPT_BUILD_NETWORK_SINK "Build Audio Sink Module (no dependencies required)" ON)
option(OPT_BUILD_NEW_PORTAUDIO_SINK "Build the new PortAudio Sink Module (Dependencies: portaudio)" OFF)

# Decoders
option(OPT_BUILD_RADIOSONDE_DECODER "Build the radiosonde decoder module (no dependencies required)" ON)
option(OPT_BUILD_ATV_DECODER "Build ATV decoder (no dependencies required)" OFF)
option(OPT_BUILD_FALCON9_DECODER "Build the falcon9 live decoder (Dependencies: ffplay)" OFF)
option(OPT_BUILD_KG_SSTV_DECODER "Build the M17 decoder module (no dependencies required)" OFF)
option(OPT_BUILD_M17_DECODER "Build the M17 decoder module (Dependencies: codec2)" OFF)
option(OPT_BUILD_METEOR_DEMODULATOR "Build the meteor demodulator module (no dependencies required)" OFF)
option(OPT_BUILD_RADIO "Main audio modulation decoder (AM, FM, SSB, etc...)" ON)
option(OPT_BUILD_WEATHER_SAT_DECODER "Build the HRPT decoder module (no dependencies required)" OFF)

# Misc
option(OPT_BUILD_BOOKMARK_MANAGER "Build the Bookmark Manager module" ON)
option(OPT_BUILD_DISCORD_PRESENCE "Build the Discord Rich Presence module" OFF)
option(OPT_BUILD_FREQUENCY_MANAGER "Build the Frequency Manager module" OFF)
option(OPT_BUILD_RECORDER "Audio and baseband recorder" OFF)
option(OPT_BUILD_RIGCTL_CLIENT "Rigctl client to make SDR++ act as a panadapter" OFF)
option(OPT_BUILD_RIGCTL_SERVER "Rigctl backend for controlling SDR++ with software like gpredict" OFF)
option(OPT_BUILD_SCANNER "Frequency scanner" OFF)
option(OPT_BUILD_SCHEDULER "Build the scheduler" OFF)

# Other options
option(USE_INTERNAL_LIBCORRECT "Use an internal version of libcorrect" ON)
option(USE_BUNDLE_DEFAULTS "Set the default resource and module directories to the right ones for a MacOS .app" OFF)

# Core of SDR++
add_subdirectory("core")

# Source modules
if (OPT_BUILD_AIRSPY_SOURCE)
add_subdirectory("source_modules/airspy_source")
endif (OPT_BUILD_AIRSPY_SOURCE)

if (OPT_BUILD_AIRSPYHF_SOURCE)
[... 133 lines cut ...]

if (OPT_BUILD_RECORDER)
add_subdirectory("misc_modules/recorder")
endif (OPT_BUILD_RECORDER)

if (OPT_BUILD_RIGCTL_CLIENT)
add_subdirectory("misc_modules/rigctl_client")
endif (OPT_BUILD_RIGCTL_CLIENT)

if (OPT_BUILD_RIGCTL_SERVER)
add_subdirectory("misc_modules/rigctl_server")
endif (OPT_BUILD_RIGCTL_SERVER)

if (OPT_BUILD_SCANNER)
add_subdirectory("misc_modules/scanner")
endif (OPT_BUILD_SCANNER)

if (OPT_BUILD_SCHEDULER)
add_subdirectory("misc_modules/scheduler")
endif (OPT_BUILD_SCHEDULER)

add_executable(sdrpp "src/main.cpp" "win32/resources.rc")
target_link_libraries(sdrpp PRIVATE sdrpp_core)

# Compiler arguments for each platform

AlexandreRouma, 2022-12-26:
>  Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so

@f5bcb Try to install this version of the API: https://www.sdrplay.com/software/SDRplay_RSP_API-ARM64-3.07.1.run
I'd recommend contacting SDRplay to tell them that their raspios version is only 32bits

> Before compiling SDR++, I modified the CMakeLists.txt file to suit my needs

@kmp56 ***NEVER*** edit the CMakeLists... Use the `-D<option>=<value>` cmake command line arguments as instructed in the readme.

f5bcb, 2022-12-27:
Hi,
The API (https://www.sdrplay.com/software/SDRplay_RSP_API-ARM64-3.07.1.run) has been installed and install script modified ( cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON).
The CMakeLists file unmodified.

I requested SDRPlay support because there is an ambiguity depending on the OS target selected on their download page: "ARM Raspberry Pi OS" or "ARM Other". I wait for a reply.

Unfortunately, I get the same problem:
....
pi@Sdrpp:~ $ sudo sdrpp --server
[2022-12-27 10:12:04.245] [info] SDR++ v1.1.0
[2022-12-27 10:12:04.246] [info] Loading config
[2022-12-27 10:12:04.252] [warning] ConfigManager locked, waiting...
[2022-12-27 10:12:04.252] [info] =====| SERVER MODE |=====
[2022-12-27 10:12:04.253] [info] Loading modules
[2022-12-27 10:12:04.253] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2022-12-27 10:12:04.256] [info] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[2022-12-27 10:12:04.258] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2022-12-27 10:12:04.260] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2022-12-27 10:12:04.262] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2022-12-27 10:12:04.270] [info] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[2022-12-27 10:12:04.272] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2022-12-27 10:12:04.275] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
**[2022-12-27 10:12:04.276] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.**
[2022-12-27 10:12:04.276] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2022-12-27 10:12:04.278] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2022-12-27 10:12:04.280] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2022-12-27 10:12:04.282] [info] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[2022-12-27 10:12:04.285] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2022-12-27 10:12:04.287] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2022-12-27 10:12:04.289] [info] Initializing Airspy Source (airspy_source)
[2022-12-27 10:12:04.296] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2022-12-27 10:12:04.303] [info] Initializing File Source (file_source)
[2022-12-27 10:12:04.303] [info] Initializing HackRF Source (hackrf_source)
[2022-12-27 10:12:04.310] [info] Initializing Hermes Source (hermes_source)
[2022-12-27 10:12:04.310] [info] Initializing PlutoSDR Source (plutosdr_source)
[2022-12-27 10:12:04.310] [info] Initializing RFspace Source (rfspace_source)
[2022-12-27 10:12:04.310] [info] Initializing RTL-SDR Source (rtl_sdr_source)
[2022-12-27 10:12:04.318] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2022-12-27 10:12:04.318] [info] Initializing SDR++ Server Source (sdrpp_server_source)
[2022-12-27 10:12:04.318] [info] Initializing SoapySDR Source (soapy_source)
[INFO] [UHD] linux; GNU C++ version 10.2.1 20201207; Boost_107400; UHD_3.15.0.0-4+b1
[2022-12-27 10:12:04.880] [info] Setting sample rate to 8000.0
[2022-12-27 10:12:04.880] [info] Initializing SpyServer Source (spyserver_source)
[2022-12-27 10:12:04.880] [info] Running post-init for Airspy Source
[2022-12-27 10:12:04.880] [info] Running post-init for AirspyHF+ Source
[2022-12-27 10:12:04.880] [info] Running post-init for File Source
[2022-12-27 10:12:04.880] [info] Running post-init for HackRF Source
[2022-12-27 10:12:04.880] [info] Running post-init for Hermes Source
[2022-12-27 10:12:04.880] [info] Running post-init for PlutoSDR Source
[2022-12-27 10:12:04.880] [info] Running post-init for RFspace Source
[2022-12-27 10:12:04.880] [info] Running post-init for RTL-SDR Source
[2022-12-27 10:12:04.880] [info] Running post-init for RTL-TCP Source
[2022-12-27 10:12:04.880] [info] Running post-init for SDR++ Server Source
[2022-12-27 10:12:04.880] [info] Running post-init for SoapySDR Source
[2022-12-27 10:12:04.880] [info] Running post-init for SpyServer Source
[2022-12-27 10:12:04.880] [info] AirspySourceModule 'Airspy Source': Menu Select!
[2022-12-27 10:12:04.881] [info] Ready, listening on 0.0.0.0:5259

f5bcb, 2022-12-27:
The SDRPLAY support response states that the 64bit API is the one in target "ARM Other"
So I used that one but the problem is still there.
SDRPP works fine with "RTL-SDR" but not with my RSP1-A
As I don't see what else I could do, I will stop wasting my time and wait for a fix.

AlexandreRouma, 2022-12-27:
There is no "fix" needed to SDR++. It's an issue on your specific system.
Likely a need to recompile with the new API installed, maybe run `sudo ldconfig`. 
Can be a lot things...

f5bcb, 2022-12-27:
I don't know what's specific about this system. It's a Rapsberry Pi 4 on which I did a fresh install of Raspbian 64 Bullseye. The API is the latest official version, the RSP1A is an original SDRPlay model and I followed the proposed Readme for the compilation and installation of SDRPP.
I don't know if such a configuration already exists and if it works with SDRPlay products.
Maybe a fix will come in the OS, in a library or something else.
Wait & see
Thanks a lot

kmp56, 2022-12-28:
What are you waiting for... I decided to check your story, and ... 100% SUCCESS! Everything compiles and works 100%. The details in the link may help you... Total install time ~3 hr.
http://www.compoz.pl/Raspberry_Pi_4_%20SDR++_%20install.pdf

AlexandreRouma, 2022-12-28:
Almost all of that information is wrong...
How about following the real manual instead?
https://www.sdrpp.com/manual.pdf

closed 2022-12-28 by AlexandreRouma

f5bcb, 2022-12-28:
Hi,

Already tested but not working : "sdrpp --server" as specified in 
documentation (p58) fail.

-    ***@***.***:~ $ sdrpp --server
     -bash: /usr/bin/sdrpp: Aucun fichier ou dossier de ce type
     ***@***.***:~ $

***@***.***:/usr/bin $ ll sdrpp*
-rwxr-xr-x 1 pi pi 10664 12 déc.  00:20 sdrpp
***@***.***:/usr/bin $

- is the "sdrpp_raspios_bullseye_armhf.deb" for 64 bits OS ?

Jeff

kmp56, 2022-12-28:
@AlexandreRouma  There are always many roads leading to the goal... You just have to choose the right one for you...

AlexandreRouma, 2022-12-30:
I'm removing all the bad information from this thread, I don't need other people having issues following bad instructions and having to deal with even more support questions...

AlexandreRouma, 2022-12-30:
@f5bcb no, sdrpp_raspios_bullseye_armhf is for 32bit raspios.

H6LS1S, 2023-08-21:
@AlexandreRouma Hello, the topic of building sdrpp for arm architectures is very poorly documented, especially when you still need to build it for the 64 bit version with SDRPlay support. 

For me personally, there is still an issue with cutting out all the unnecessary to get a lightweight package from sdrpp server + sdrplay source, I experimented a lot with options cmake, but when I start the server, I still see that unnecessary modules are installed

The following build scenario worked for me on my Raspberry 4:
```bash

sudo apt install -y build-essential cmake git p7zip-full wget \
 libfftw3-dev libglfw3-dev libglew-dev libvolk2-dev libsoapysdr-dev \
 libairspyhf-dev libairspy-dev libiio-dev libad9361-dev librtaudio-dev \
 libhackrf-dev librtlsdr-dev libbladerf-dev liblimesuite-dev libzstd1 

wget https://www.sdrplay.com/software/SDRplay_RPi_Scripts_v0.3.zip
unzip SDRplay_RPi_Scripts_v0.3.zip && sh 1installAPI.sh

git clone https://github.com/AlexandreRouma/SDRPlusPlus
cd SDRPlusPlus && mkdir build && cd build 
cmake .. -DOPT_BUILD_SDRPLAY_SOURCE=ON && make VERBOSE=1
  
cd .. \
  && sh make_debian_package.sh ./build 'libfftw3-dev, libglfw3-dev, libvolk2-dev, librtaudio-dev'  \
  && mv sdrpp_debian_amd64.deb sdrpp_debian_armhf.deb
  && sudo dpkg -i sdrpp_debian_armhf.deb
  
sdrpp --server --port 5656
```

in fact, this is a hodgepodge from the README and Github Aaction, and I hope this hodgepodge will help someone ))

AlexandreRouma, 2023-09-29:
Don't use a premade script, follow the linux instructions, there is no difference between architectures.