AlexandreRouma/SDRPlusPlus #947: Audio routing improvements

kord1e, 2022-12-26:
**Feature description**
Would be useful to have some more options for audio routing:

1) Choose to which system audio interface/sink to send audio from SDR++ audio sink. This would be useful with some digital decoders, for example the Dream DRM decoder.
Gqrx has this feature (it's a bit impractical, not very flexible, SDR++ could do it even better)  and you can demodulate a DRM station with USB, create a virtual audio sink in pulseaudio, choose to output audio from Gqrx to virtual sink and input it in Dream from virtual sink. Then the USB demodulated stuff doesn't go to speakers, only the decoded DRM radio does.

2) Choose which channel to output audio (mono) from SDR++ audio sink. For example have two demodulators active and output one to left and the other to right channel if you have stereo system.

AlexandreRouma, 2022-12-26:
1) Has always been possible, just chose the device in the Sink menu for the particular VFO.
2) Duplicate of #653 but will likely not be implemented any time soon as extremely few users need or ever want it.

closed 2022-12-26 by AlexandreRouma

kord1e, 2022-12-26:
@AlexandreRouma 
Ok, thanks,

to 1), SDR++ only shows these options in the sinks menu:
- in the first menu, there are "None", "Audio" and "Network" (the network option is nice but can't be used with all decoding software)
- in the second menu, there is just "PulseAudio" when Audio is selected in the first menu, nothing else
- the third menu is the sample rate

So there's no option for the pulseaudio sink.

I'm using Debian at the moment and SDR++ was compiled a couple of weeks ago from master branch.

AlexandreRouma, 2022-12-26:
that's the sink type, not the sink device.
The actual device is the option under that.

kord1e, 2022-12-26:
@AlexandreRouma 
Hm, what am I doing wrong? I don't see any extra settings here. The second menu doesn't have any options except "PulseAudio". No way to choose which Pulseaudio sink is used.

[image]

kord1e, 2022-12-26:
For comparison, in Gqrx you can choose the sink here:

[image]

AlexandreRouma, 2022-12-26:
You likely have some other audio server on your system and RTaudio is using that instead, not giving access to each pulseaudio sink.

This fix (althought for another issue) may help: #112

kord1e, 2022-12-26:
Ok, looks like I have both pipewire and pulseaudio running. Maybe that's why? 

Removing pulseaudio wouldn't even be a big issue, but Gqrx package requires it so it would be removed as well. So this must be sorted out with Gqrx next, I think.

AlexandreRouma, 2022-12-26:
If you're on a recent ubuntu version pipewire is mandatory.
The fix I sent will just force RTaudio to use pulse and not pipewire.

kord1e, 2022-12-27:
@AlexandreRouma 
I'm using Debian so pipewire is not yet required by the distro... 
I removed pipewire but there's still only Pulseaudio in that menu.
I'll try to find something in audio configs or use the patched RTaudio, thanks...