AlexandreRouma/SDRPlusPlus #948: Performance issue

drozdi70, 2022-12-29:
Hi
I have just installed SDR++ on my raspaberry 4 Pi B (4GB RAM CPU 1800Mhz) and it seems it works ok but them main issue is that it consumes 80-90% of CPU :(. Why it takes so much CPUs even only launched? Any idea how it could be reduced?
Version 1.1 built and compiled from git repository.
Kindly

AlexandreRouma, 2022-12-29:
You're running OpenGL software rendered instead of on the GPU, check your graphics drivers.
I can't help more than that, I don't have a raspberry pi to test anything but other users have it running normally with no issue.

closed 2022-12-29 by AlexandreRouma

drozdi70, 2022-12-29:
how it could be tested that you are right?I can make tests by myself Temat: Re: [AlexandreRouma/SDRPlusPlus] Performance issue (Issue #948)Data: 2022-12-29 20:12Nadawca: "AlexandreRouma" ***@***.***&gt;Adresat: "AlexandreRouma/SDRPlusPlus" ***@***.***&gt;; DW: "drozdi70" ***@***.***&gt;; "Author" ***@***.***&gt;; 

You're running OpenGL software rendered instead of on the GPU, check your graphics drivers.
I can't help more than that, I don't have a raspberry pi to test anything but other users have it running normally with no issue.