AlexandreRouma/SDRPlusPlus #953: Only radio and SDR++ server

iphands, 2023-01-03:
I would like to be able to build sdr++ without building the GUI at all.
I really just want the minimum needed to operate the my hardware and setup an SDR++ Server.

I was playing around with the `OPT_BUILD*` options but I dont think any allow for this behavior.
Could we have some minimal build option or a _OPR_BUILD_NOGUI or something?
Perhaps we could even have a make target that builds `sdrpp_server` as a standalone binary.

AlexandreRouma, 2023-01-14:
Literally no point. In server mode graphics stuff isn't loaded and doesn't use up memory. And the executable file size of like 2mb is negligeable on any PC from the last 30 years.

closed 2023-01-14 by AlexandreRouma

iphands, 2023-01-15:
@AlexandreRouma There is a point though... On some devices its a pita to even build all that stuff.
On an RPI3 you have to build with 1 or 2 parallel jobs or you oom to death.