AlexandreRouma/SDRPlusPlus #955: When using Source:SDR++ Server the Source[REMOTE] no longer has DB info or Label displayed on gain control

iphands, 2023-01-04:
**Hardware**
Radio: RTL-SDR v3 (https://rtl-sdr.com/v3)
Computers n/a tested on multiple machine combinations:
- RPI3 server -> Linux Desktop client
- MacOS server -> Linux Desktop client
- RPI3 server -> Linux Laptop client
- Linux Desktop server -> (same) Linux Desktop client

**Software**
- Operating System: Various (OSX, Gentoo, Fedora 37)
- SDR++:
```
commit a1cbc69a65b1d36149d45910f57dc881148d9808 (HEAD -> master, origin/master, origin/HEAD)
Author: AlexandreRouma <alexandre.rouma@gmail.com>
Date:   Mon Jan 2 23:01:45 2023 +0100

    temporarily removed the ignore silence option since it doesn't work properly
```

**Bug Description**
See image. When I use the USB RTL-SDR locally (no sdrpp server in between) I can see a "Gain" label and value (in db). When I use SRD++ server and the same USB HW as the remote source... I can still control the gain but I see no label/value. Just an unlabeled slider with no DB value to look at.

**Steps To Reproduce**
1. Run an `sdrpp --sever`
2. Connect to server via SRD++ Server source
3. Start stream
4. Observe gain control with no label/val

**Screenshots**
[image]

AlexandreRouma, 2023-01-04:
Not a bug, just a limitation of the remote system. It is switched to a blank slider on purpose.

closed 2023-01-04 by AlexandreRouma

iphands, 2023-01-05:
@AlexandreRouma can we
1) Add the label without the value? That should not be a limitation of passing around data eh?
2) Have adding the value as a feature request instead of a bug??

In the end... Bug or feature classification aside I think having this information be passed along and rendered on the client side is desired user experience.

AlexandreRouma, 2023-01-05:
1. The label is in neither the server mode nor the local mode. I didn't put it because it's the only slider in there, the purpose was pretty obvious. I guess it can be added to be more consistent with the other ones.

2. Adding the value is an absolute pain. If it was easy I would have done it when the server mode was released. If you look at the code it's clearly commented that not showing the value is non ideal but necessary for now. The remote system wasn't made to have dynamically changing text. The RTL-SDR's gain setting doesn't work like other SDRs. it's non linear. You can't just display the slider value like all the other sources.

AlexandreRouma, 2023-01-05:
the value stored in the config is not the dB value. It's just an index that selects a certain gain from the list given by librtlsdr. Having it as a slider gives the illusion of being continous but it's not.

iphands, 2023-01-05:
> the value stored in the config is not the dB value. It's just an index that selects a certain gain from the list given by librtlsdr. Having it as a slider gives the illusion of being continous but it's not.

Ahh I see. So when I operate locally, the displayed DB is the result of some real data coming back from librtlsdr or not?
Or am I imagining that the slider had gain info locally. I might be.

iphands, 2023-01-05:
> Or am I imagining that the slider had gain info locally. I might be.
Oh could it be that when I first set this up I was on a windows VM?
ie does the slider have gain info on windows but not on *nix?

Just hit me that this may be the case.

AlexandreRouma, 2023-01-05:
> the displayed DB is the result of some real data coming back from librtlsdr or not?

yes

> Oh could it be that when I first set this up and was running local I was on a windows VM?

It displays the gain in dB but doesn't have a label that says "gain" next to the slider.

iphands, 2023-01-05:
> It displays the gain in dB but doesn't have a label that says "gain" next to the slider.

Oh I see, so locally, no label but value.
Remote no label no value.

I think that is where my brain got goofed. Thanks.