AlexandreRouma/SDRPlusPlus #956: SDRpp error Could not list devices -7 and Could not intiatialized the SDRplay API. Make sure that the service is running.

TTorreS96, 2023-01-04:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: Intel Core i5 650
- RAM: 8GB 
- GPU: NA
- SDR: local

**Software**
- Operating Windows 10 x64
- SDR++: SDR++ Nightly Build

**Bug Description**
The program opens and closes immediately.
Running from the command line gives these errors "Could not list devices -7" and "Could not intiatialized the SDRplay API. Make sure that the service is running."

**Steps To Reproduce**
1. Open the program

**Additional info**
C:\Users\Jose Torres>C:\sdrpp_windows_x64\sdrpp.exe -s
[2023-01-04 21:34:02.062] [info] SDR++ v1.1.0
[2023-01-04 21:34:02.064] [info] Loading config
[2023-01-04 21:34:02.066] [info] =====| SERVER MODE |=====
[2023-01-04 21:34:02.066] [info] Loading modules
[2023-01-04 21:34:02.067] [info] Loading C:/Users/Jose Torres/modules/airspyhf_source.dll
[2023-01-04 21:34:02.070] [info] Loading C:/Users/Jose Torres/modules/airspy_source.dll
[2023-01-04 21:34:02.071] [info] Loading C:/Users/Jose Torres/modules/bladerf_source.dll
[2023-01-04 21:34:02.075] [info] Loading C:/Users/Jose Torres/modules/file_source.dll
[2023-01-04 21:34:02.078] [info] Loading C:/Users/Jose Torres/modules/hackrf_source.dll
[2023-01-04 21:34:02.080] [info] Loading C:/Users/Jose Torres/modules/hermes_source.dll
[2023-01-04 21:34:02.081] [info] Loading C:/Users/Jose Torres/modules/limesdr_source.dll
[2023-01-04 21:34:02.083] [info] Loading C:/Users/Jose Torres/modules/plutosdr_source.dll
[2023-01-04 21:34:02.085] [info] Loading C:/Users/Jose Torres/modules/rfspace_source.dll
[2023-01-04 21:34:02.087] [info] Loading C:/Users/Jose Torres/modules/rtl_sdr_source.dll
[2023-01-04 21:34:02.088] [info] Loading C:/Users/Jose Torres/modules/rtl_tcp_source.dll
[2023-01-04 21:34:02.090] [info] Loading C:/Users/Jose Torres/modules/sdrplay_source.dll
[2023-01-04 21:34:02.094] [info] Loading C:/Users/Jose Torres/modules/sdrpp_server_source.dll
[2023-01-04 21:34:02.096] [info] Loading C:/Users/Jose Torres/modules/soapy_source.dll
[2023-01-04 21:34:02.098] [info] Loading C:/Users/Jose Torres/modules/spyserver_source.dll
[2023-01-04 21:34:02.100] [info] Initializing Airspy Source (airspy_source)
[2023-01-04 21:34:02.132] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2023-01-04 21:34:02.162] [info] Initializing BladeRF Source (bladerf_source)
[2023-01-04 21:34:02.192] [error] Could not list devices -7
[2023-01-04 21:34:02.193] [info] Initializing File Source (file_source)
[2023-01-04 21:34:02.194] [info] Initializing HackRF Source (hackrf_source)
[2023-01-04 21:34:02.222] [info] Initializing Hermes Source (hermes_source)
[2023-01-04 21:34:02.222] [info] Initializing LimeSDR Source (limesdr_source)
[2023-01-04 21:34:02.223] [info] Initializing PlutoSDR Source (plutosdr_source)
[2023-01-04 21:34:02.224] [info] Initializing RFspace Source (rfspace_source)
[2023-01-04 21:34:02.224] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R820T tuner
[2023-01-04 21:34:02.813] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2023-01-04 21:34:02.815] [info] Initializing SDR++ Server Source (sdrpp_server_source)
[2023-01-04 21:34:02.816] [info] Initializing SDRplay Source (sdrplay_source)
[2023-01-04 21:34:02.819] [error] Could not intiatialized the SDRplay API. Make sure that the service is running.
[2023-01-04 21:34:02.819] [info] Initializing SoapySDR Source (soapy_source)
←[1m←[31m[ERROR] SoapySDR::loadModule(C:\Program Files\PothosSDR/lib/SoapySDR/modules0.8/volkConverters.dll)
  LoadLibrary() failed: ImpossÝvel localizar o procedimento especificado.
←[0m
←[1m←[31m[ERROR] sdrplay_api_Open() Error: sdrplay_api_Fail←[0m
←[1m←[31m[ERROR] Please check the sdrplay_api service to make sure it is up. If it is up, please restart it.←[0m
[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107500; UHD_4.0.0.HEAD-0-g90ce6062
Found Rafael Micro R820T tuner
←[1m←[31m[ERROR] SoapySDR::Device::enumerate(sdrplay) sdrplay_api_Open() failed←[0m
[INFO] Opening Generic RTL2832U OEM :: 00000001...
Found Rafael Micro R820T tuner
[2023-01-04 21:34:05.726] [info] Setting sample rate to 250000.0
[2023-01-04 21:34:05.727] [info] Initializing SpyServer Source (spyserver_source)
[2023-01-04 21:34:05.729] [info] Running post-init for Airspy Source
[2023-01-04 21:34:05.731] [info] Running post-init for AirspyHF+ Source
[2023-01-04 21:34:05.732] [info] Running post-init for BladeRF Source
[2023-01-04 21:34:05.733] [info] Running post-init for File Source
[2023-01-04 21:34:05.734] [info] Running post-init for HackRF Source
[2023-01-04 21:34:05.739] [info] Running post-init for Hermes Source
[2023-01-04 21:34:05.741] [info] Running post-init for LimeSDR Source
[2023-01-04 21:34:05.742] [info] Running post-init for PlutoSDR Source
[2023-01-04 21:34:05.744] [info] Running post-init for RFspace Source
[2023-01-04 21:34:05.745] [info] Running post-init for RTL-SDR Source
[2023-01-04 21:34:05.747] [info] Running post-init for RTL-TCP Source
[2023-01-04 21:34:05.748] [info] Running post-init for SDR++ Server Source
[2023-01-04 21:34:05.749] [info] Running post-init for SDRplay Source
[2023-01-04 21:34:05.755] [info] Running post-init for SoapySDR Source
[2023-01-04 21:34:05.756] [info] Running post-init for SpyServer Source
[2023-01-04 21:34:05.758] [info] AirspySourceModule 'Airspy Source': Menu Select!
[2023-01-04 21:34:05.764] [info] Ready, listening on 0.0.0.0:5259

krutzy62, 2023-01-06:
I get a similar error with the Jammy Ubuntu version centering on the sdrplay load.  It has the same error "Could not list devices -7"

[2023-01-06 00:16:28.877] [info] SDR++ v1.1.0
[2023-01-06 00:16:28.878] [info] Loading config
[2023-01-06 00:16:29.273] [info] Using OpenGL 3.0
[2023-01-06 00:16:29.465] [info] Loading icons
[2023-01-06 00:16:29.486] [info] Loading band plans
[2023-01-06 00:16:29.531] [info] Loading band plans color table
[2023-01-06 00:16:29.542] [error] Menu element is missing name key
[2023-01-06 00:16:29.559] [info] Loading modules
[2023-01-06 00:16:29.561] [info] Loading /usr/lib/sdrpp/plugins/network_sink.so
[2023-01-06 00:16:29.587] [info] Loading /usr/lib/sdrpp/plugins/file_source.so
[2023-01-06 00:16:29.599] [info] Loading /usr/lib/sdrpp/plugins/discord_integration.so
[2023-01-06 00:16:29.625] [info] Loading /usr/lib/sdrpp/plugins/new_portaudio_sink.so
[2023-01-06 00:16:29.641] [info] Loading /usr/lib/sdrpp/plugins/bladerf_source.so
[2023-01-06 00:16:29.668] [info] Loading /usr/lib/sdrpp/plugins/audio_sink.so
[2023-01-06 00:16:29.693] [info] Loading /usr/lib/sdrpp/plugins/airspy_source.so
[2023-01-06 00:16:29.702] [info] Loading /usr/lib/sdrpp/plugins/m17_decoder.so
[2023-01-06 00:16:29.722] [info] Loading /usr/lib/sdrpp/plugins/spyserver_source.so
[2023-01-06 00:16:29.727] [info] Loading /usr/lib/sdrpp/plugins/soapy_source.so
[2023-01-06 00:16:29.746] [info] Loading /usr/lib/sdrpp/plugins/rigctl_server.so
[2023-01-06 00:16:29.755] [info] Loading /usr/lib/sdrpp/plugins/rtl_sdr_source.so
[2023-01-06 00:16:29.767] [info] Loading /usr/lib/sdrpp/plugins/scanner.so
[2023-01-06 00:16:29.779] [info] Loading /usr/lib/sdrpp/plugins/frequency_manager.so
[2023-01-06 00:16:29.795] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2023-01-06 00:16:29.808] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2023-01-06 00:16:29.822] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2023-01-06 00:16:29.835] [info] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[2023-01-06 00:16:29.863] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2023-01-06 00:16:29.877] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2023-01-06 00:16:29.890] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2023-01-06 00:16:29.912] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2023-01-06 00:16:29.928] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2023-01-06 00:16:29.935] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[2023-01-06 00:16:29.935] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2023-01-06 00:16:29.951] [info] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[2023-01-06 00:16:29.970] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so
[2023-01-06 00:16:29.994] [info] Loading /usr/lib/sdrpp/plugins/sdrpp_server_source.so
[2023-01-06 00:16:30.004] [info] Initializing Airspy Source (airspy_source)
[2023-01-06 00:16:30.018] [info] Initializing AirspyHF+ Source (airspyhf_source)
[2023-01-06 00:16:30.030] [info] Initializing Audio Sink (audio_sink)
[2023-01-06 00:16:30.041] [info] Initializing BladeRF Source (bladerf_source)
[2023-01-06 00:16:30.058] [error] Could not list devices -7
[2023-01-06 00:16:30.058] [info] Initializing File Source (file_source)
[2023-01-06 00:16:30.062] [info] Initializing Frequency Manager (frequency_manager)
[2023-01-06 00:16:30.076] [info] Initializing HackRF Source (hackrf_source)
[2023-01-06 00:16:30.092] [info] Initializing Hermes Source (hermes_source)
[2023-01-06 00:16:30.109] [info] Initializing LimeSDR Source (limesdr_source)
[2023-01-06 00:16:30.146] [info] Initializing Network Sink (network_sink)
[2023-01-06 00:16:30.148] [info] Initializing PlutoSDR Source (plutosdr_source)
[2023-01-06 00:16:30.162] [info] Initializing RFspace Source (rfspace_source)
[2023-01-06 00:16:30.173] [info] Initializing RTL-SDR Source (rtl_sdr_source)
Found Rafael Micro R820T tuner
[2023-01-06 00:16:30.655] [info] Initializing RTL-TCP Source (rtl_tcp_source)
[2023-01-06 00:16:30.658] [info] Initializing Radio (radio)
[Resamp] predec: 32, interp: 4, decim: 5, inacc: 0.000000%, taps: 380
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 32, interp: 5000, decim: 7813, inacc: 0.000000%, taps: 593750
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[Resamp] predec: 4, interp: 96, decim: 125, inacc: 0.000000%, taps: 9500
[2023-01-06 00:16:30.850] [warning] Demod switch took 166416 us
[2023-01-06 00:16:30.857] [info] RtAudio stream open
[2023-01-06 00:16:30.857] [info] Initializing Recorder (recorder)
[2023-01-06 00:16:30.858] [info] Initializing Rigctl Server (rigctl_server)
[2023-01-06 00:16:30.860] [info] Initializing SDR++ Server Source (sdrpp_server_source)
[2023-01-06 00:16:30.864] [info] Initializing SDRplay Source (sdrplay_source)
[2023-01-06 00:16:30.868] [error] Module 'sdrplay_source' doesn't exist
[2023-01-06 00:16:30.868] [info] Initializing SoapySDR Source (soapy_source)
Illegal instruction (core dumped)

krutzy62, 2023-01-06:
I don't know if this helps but if you compare the following lines, I think the problem is occurring because there is a stray period (.) at the end of the sdrplay load.  It is the only one that is trying to load with *.so. not *.so

Maybe that is all it is.

[2023-01-06 18:30:08.024] [info] Loading /usr/lib/sdrpp/plugins/recorder.so
[2023-01-06 18:30:08.038] [info] Loading /usr/lib/sdrpp/plugins/plutosdr_source.so
[2023-01-06 18:30:08.049] [info] Loading /usr/lib/sdrpp/plugins/hackrf_source.so
[2023-01-06 18:30:08.059] [info] Loading /usr/lib/sdrpp/plugins/hermes_source.so
[2023-01-06 18:30:08.082] [info] Loading /usr/lib/sdrpp/plugins/rfspace_source.so
[2023-01-06 18:30:08.100] [info] Loading /usr/lib/sdrpp/plugins/rtl_tcp_source.so
[2023-01-06 18:30:08.125] [info] Loading /usr/lib/sdrpp/plugins/radio.so
[2023-01-06 18:30:08.137] [info] Loading /usr/lib/sdrpp/plugins/limesdr_source.so
[2023-01-06 18:30:08.150] [info] Loading /usr/lib/sdrpp/plugins/sdrplay_source.so
[2023-01-06 18:30:08.160] [error] Couldn't load /usr/lib/sdrpp/plugins/sdrplay_source.so.
[2023-01-06 18:30:08.160] [info] Loading /usr/lib/sdrpp/plugins/meteor_demodulator.so
[2023-01-06 18:30:08.178] [info] Loading /usr/lib/sdrpp/plugins/spectran_http_source.so
[2023-01-06 18:30:08.187] [info] Loading /usr/lib/sdrpp/plugins/airspyhf_source.so

evoke0, 2023-01-10:
#463 is related, looks like a problem with SDRPlay's API. FYI: I tried running sdrpp as server mode and got the same 2 errors, but still runs on CLI, although nothing is served on 0.0.0.0:5259, just a blank page trying to load. Launching the GUI shows the same output but works as intended. Ubuntu Jammy nightly build

krutzy62, 2023-01-10:
Ok - so I need to start SDRPlay service before running SDR++ even if I don't have an SDRPlay device, only having an RTL-SDR?

AlexandreRouma, 2023-01-14:
It's almost surement a graphics driver issue. SDR++ *requires* OpenGL 2.1 minimum.

krutzy62, 2023-01-15:
Very well.  I will look into that.
Thank you.

krutzy62, 2023-01-19:
I looked at my system and according to: glxinfo | grep "OpenGL version"
My result is: OpenGL version string: 3.0 Mesa 22.0.5
Should that suffice?

AlexandreRouma, 2023-01-22:
@krutzy62 I was responding to the original poster of the issue @TTorreS96.
Yours is completely different. Just uninstall SoapySDR if you don't need it or edit the ~/.config/sdrpp/config.json to remove this exact part:

```
        "SoapySDR Source": {
            "enabled": true,
            "module": "soapy_source"
        },
```

krutzy62, 2023-01-23:
Thank you sir!  That did it.

 

Kevin

closed 2023-03-10 by AlexandreRouma