AlexandreRouma/SDRPlusPlus #958: Higher bandwidth for demodulators

kord1e, 2023-01-07:
**Feature description**

Currently SDR++ allows quite limited bandwidth for AM/SSB/DSB demodulators. It's often enough but sometimes there are signals that would benefit from higher bandwidth. For example, Gqrx allows higher bandwidth for these modes compared to SDR++.

Are there disadvantages if higher bandwidth is allowed?

AlexandreRouma, 2023-01-07:
> Are there disadvantages if higher bandwidth is allowed?

Wasting CPU usage by running the demodulators at a higher samplerate for no reason or dramatically increasing complexity and and decreasing responsiveness by dynamically changing the samplerate at which the demodulators run?

There is no reason to have the modes go wider than actually needed because they are configured for normal usage. If you're trying to receive a complete out of spec signal it likely makes much more sense to have a dedicated demodulator.

closed 2023-01-07 by AlexandreRouma

kord1e, 2023-01-08:
@AlexandreRouma 
Ok, I see. But SDR++ already performs better than other SDR software so is it much resources wasted? :P