AlexandreRouma/SDRPlusPlus #962: UHD support?

dengzi118, 2023-01-12:
**Feature description**
i can‘t find uhd when I find device  i can‘t find uhd 
Is my operation wrong？

AlexandreRouma, 2023-01-12:
As described in the module list of the readme, the USRP source is not built in releases. You must build from source with the appropriate cmake flags.

closed 2023-01-12 by AlexandreRouma

Josh-SMU, 2023-07-29:
I can't for the life of me get this to work. Trying to run from root_dev with modules manually populated returns:

```
[INFO] [UHD] linux; GNU C++ version 12.3.0; Boost_107400; UHD_4.4.0.0+ds1-4
[WARNING] [B200] EnvironmentError: IOError: Could not find path for image: usrp_b200_fw.hex

Using images directory: <no images directory located>

Set the environment variable 'UHD_IMAGES_DIR' appropriately or follow the below instructions to download the images package.

Please run:

 "/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"
 ```
 
 But this file does not exist.

AlexandreRouma, 2023-07-29:
This has nothing to do with SDR++.https://kb.ettus.com/B200/B210/B200mini/B205mini_Getting_Started_Guides#Install_and_Setup_the_Software_Tools_on_Your_Host_Computer

Josh-SMU, 2023-08-01:
Thank you, I got it working, however the UHD option is still missing from SDR++. I was able to confirm the Ettus works with other scripts, but following the instructions on building SDR++ from source, force enabling the USRP driver, modifying the config.json in root_dev, and even in the installed copy, have all failed to add the option to the receiver list. What am I missing?