AlexandreRouma/SDRPlusPlus #967: Recorder plugin doesn't show "Ignore Silence"

weldp, 2023-01-18:
FIRST: Before reporting any bug, make sure that the bug you are reporting has not been reported before. Also, try to use the [nightly version](https://www.sdrpp.org/nightly) if possible in case I've already fixed the bug.

**Hardware**
- CPU: AMD FX-8320
- RAM: 8GB
- GPU: NVIDIA GeForce RTX 3060 Ti
- SDR: NESDR SMArt RTLSDR

**Software**
- Operating System: Windows 10 x64
- SDR++: 1.10 + 22:04:22, Jan 13, 2023

**Bug Description**
When looking at the settings for the recorder plugin there is no setting for "Ignore Silence" like it says in the manual. Instead it shows "Stereo" and has a checkbox for that. I noted that it didn't have an option for stereo in the manual.

**Steps To Reproduce**
1. Add module for recorder plugin
2. Name it recorder
3. Go to settings for recorder and see that there is no setting for "Ignore Silence" but there is one for "Stereo"
4.  I tried checking "Stereo" to see if that would instead ignore silence but it still recorded silence. 

[image]

ericek111, 2023-01-18:
Temporarily removed here: https://github.com/AlexandreRouma/SDRPlusPlus/commit/a1cbc69a65b1d36149d45910f57dc881148d9808

I believe it didn't work in the old recorder either.

AlexandreRouma, 2023-01-18:
yeah not a bug, just removing useless stuff.

closed 2023-01-18 by AlexandreRouma